ETLCPP/etl #607: Feature request: add select1st and select2nd functors

evantill, 2022-09-16:
What do you think of adding [select1st](https://www.boost.org/sgi/stl/select1st.html) and [select2nd](https://www.boost.org/sgi/stl/select2nd.html) functors in ETL library ?

_(I can submit a PR)_

jwellbelove, 2022-09-16:
Sounds like a nice and simple addition.
If you do a PR, don't forget to use the `ETL_OR_STD::pair` type.

evantill, 2022-09-20:
I have found two implementations :

- Option A:
```c++
    template <class FirstType>
    struct select1st
    {
      typedef FirstType type;

      template <class T>
      inline const type& operator()(const T& x) const
      {
        return x.first;
      }

      template <class T>
      inline type& operator()(T& x)
      {
        return const_cast<type&>(x.first);
      }
    };
```

- Option B:
```c++
    template<typename T>
    struct Select1st {
      typedef typename T::first_type type;

      inline type& operator()(T& x) const
      {
        return x.first;
      }

      inline const type& operator()(const T& x) const
      {
        return x.first;
      }
    };
```

My preferred option is option B.

Which one to use in ETL ?

jwellbelove, 2022-09-20:
I prefer option B.
Option A looks odd as it seems to require type information that was already implied in the function name.

evantill, 2022-09-20:
Perfect _(I also prefer option B)_.

evantill, 2022-10-17:
@jwellbelove can we close this issue ?

jwellbelove, 2022-10-17:
To be included in next release.

closed 2022-10-17 by jwellbelove