ETLCPP/etl #608: Type is_trivially_copyable differs from std

jputcu, 2022-09-18:
Wanting to use `etl::bit_cast` my type has to be `trivially_copyable`, but using the `ETL_NO_STL` flags, which is needed on AVR, I get different results:
~~~~
struct EncodedData { char buf[3]; }
~~~~

Using `static_assert(etl::is_trivially_copyable<EncodedData>::value)` without `ETL_NO_STL`  is ok. With or on AVR it fails.

See demonstration at https://godbolt.org/z/4c9z9rTvP

jputcu, 2022-09-18:
Related to the `bit_cast`, there is a typo in the documentation of `bit_cast`: the function argument should be of type `TSource`

jwellbelove, 2022-09-18:
Without the STL it is difficult to implement some traits as they require the use of compiler specific 'built-ins', which may, or may not, exist on a particular version of a compiler.

You could try defining `ETL_USE_TYPE_TRAITS_BUILTINS` and see if that works for AVR.

jwellbelove, 2022-09-18:
Another option is to define `ETL_USER_DEFINED_TYPE_TRAITS` and declare the traits yourself.

See [https://www.etlcpp.com/type_traits.html](https://www.etlcpp.com/type_traits.html)

jputcu, 2022-09-18:
I did a small test with `ETL_USE_TYPE_TRAITS_BUILTINS` on `AVR` and it appears to work: https://godbolt.org/z/53PcffPoK.
Then, perhaps the question is if we can autodetect more defines, I already have `-DETL_TARGET_DEVICE_GENERIC -DETL_TARGET_OS_NONE -DETL_NO_STL -DETL_NO_CPP_NAN_SUPPORT -DETL_NO_HUGE_VAL_SUPPORT -DETL_USE_LEGACY_VARIANT`?
I mostly thinking about reducing the threshold for newcomers to use the wonderful `etl`.

jwellbelove, 2022-09-18:
I've realised that I started to go down the 'auto detect' route a while ago.
There are some macros and ETL traits that are defined using `__has_builtin` (if available), but they are not used at present.

e.g.
`ETL_USING_BUILTIN_IS_TRIVIALLY_COPYABLE` (`0` or `1`)
and
`etl::traits::using_builtin_is_trivially_copyable` (`false` or `true`)

jwellbelove, 2022-09-20:
I'll put automatic detection of builtins for type traits on my TODO list.

closed 2022-09-20 by jwellbelove