ETLCPP/etl #611: string typedefs

daniel-brosche, 2022-09-23:
my proposal is to introduce some etl::string-typedefs with the size of power of two, to simplify common usage.

```cpp
    typedef string<8>  string8;
    typedef string<16>  string16;
    typedef string<32>  string32;
    typedef string<64>  string64;
    typedef string<128>  string128;
    typedef string<256>  string256;
    typedef string<512>  string512;
    typedef string<1024>  string1024;
    typedef string<2048>  string2048;
    typedef string<4096>  string4096;
    typedef string<8192>  string8192;
    typedef string<16384> string16384;

```

daniel-brosche, 2022-10-10:
I have noticed that in my common scenarios I often need to use strings up to 4096 characters. It is tedious to write etl::string<4096> because the autocompletion does only provide etl::string and the rest has to be done manualy or by copy and paste. 
So I started to create typedefs for the most common sized strings and did put them in the etl namespace. 
This typedefs can be reused.
In my opinion it simplifies writing.

jwellbelove, 2022-10-10:
I think this is a solution to a problem that is only specific to a very small number of people.
I think the best bet would be for you to put these in your own namespace.

daniel-brosche, 2022-10-10:
Ok, it was just an proposal

closed 2022-10-10 by daniel-brosche

jwellbelove, 2022-10-10:
Thanks for suggesting anyway.
All proposals are welcome. :-)