ETLCPP/etl #615: Feature request: addition of etl::expected

jputcu, 2022-10-02:
Could it be possible to add `etl::expected`, matching the [c++23 std::expected](https://en.cppreference.com/w/cpp/utility/expected).
Because the lack of exceptions on my AVR platform, I have used `etl::variant` for this, but having a more standard way would be better.
This could be a replacement of `etl::result`.

jwellbelove, 2022-10-02:
Thanks, I'll take a look at that

jwellbelove, 2022-11-02:
Implementation of `etl::unexpected` in 20.35.0

jputcu, 2023-02-06:
I was planning to use this, but noticed only `unexpected` was provided.
Looking around I found two nice, active, alternatives:

1. https://github.com/martinmoene/expected-lite
2. https://github.com/TartanLlama/expected

To not reinvent the wheel it would be nice if we could use them, but I quickly run into `AVR` limitations, e.g. missing `utility` header. Although `etl` probably provides everything required.
Could it be possible to provide a layer to satisfy the `std` library?

This layer would allow more lightweight C++ libraries to be used, e.g. `https://github.com/boost-ext/sml`.

jwellbelove, 2023-02-06:
The trouble with the last thing I said about headers is that the compiler would not be able to distinguish which template library (STL or ETL) to use when it sees something like `#include <algorithm>`.

jputcu, 2023-02-06:
I think all `etl` headers end with `.h`, so we could perhaps a directory with the official headers which forward to the `.h` headers and alias the namespace?

jwellbelove, 2023-02-06:
I think the problem is that if you did that, the compiler will will probably still find its own headers before the ETL's.

`#include <algorithm>` <<< I think the compiler will choose its own <algorithm> header first.

jwellbelove, 2023-02-06:
With GCC it seems you can use `-nostdinc++` as a compiler option. I don't know about other compilers.

jwellbelove, 2023-02-23:
Fixed 20.35.12

closed 2023-02-23 by jwellbelove