ETLCPP/etl #626: Empty `etl::optional` ctor storage initialization performance

mkalibat, 2022-10-18:
Hi,

[This PR](https://github.com/ETLCPP/etl/pull/381/) added initialization of `etl::optional`'s storage buffer in the empty-state constructors to avoid compiler warnings.

A problem with this is that now constructing optionals in the empty state requires value initialization (rather than trivial default construction, which "performs no action") of the storage buffer to 0's, which can be quite inefficient in some cases. (See [Explanation 3](https://en.cppreference.com/w/cpp/language/value_initialization) and [Trivial default constructor](https://en.cppreference.com/w/cpp/language/default_constructor)):

See https://godbolt.org/z/jMd59sc9T.

Looking at the `std::optional` implementations, it looks like some zero-initialize the storage and some don't.

I'm not sure what the best solution is here; being warning-clean is essential, but the extra overhead of the zero-init can be substantial. In my case, this causes ~10% CPU overhead during fairly frequently executed operations when using a not-very-large optional (~36B). I can work around it locally, but it may be a surprising performance hit for users in general. I suppose the `[[maybe_unused]]` attribute might be used to suppress the warnings, but that's C++17.

Any thoughts? Thanks.

jwellbelove, 2022-10-19:
I think there are a couple of other classes that do the same as etl::optional to suppress warnings. I may need to disable that warning for the constructors.

mkalibat, 2022-10-19:
Great, thanks for taking a look!

jwellbelove, 2022-11-02:
Fixed 20.35.0

closed 2022-11-02 by jwellbelove