ETLCPP/etl #629: Open-cmsis package

Jonny-vb, 2022-10-31:
Would you be interested in making an open-cmsis package for ETL?

https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/index.html
https://developer.arm.com/tools-and-software/embedded/cmsis/cmsis-packs

If you're not familiar with them, they're used to pull in software components for ARM chips. I believe they originally came out of uVision, but have support in some other IDEs too now. (I think there's a reasonable effort ongoing currently to improve the whole ecosystem around the cmsis usage currently).

Hopefully the benefit to you would be greater discoverability for the project (converted to sponsorships...).

The packs can become quite complicated with regards to setting up dependencies, but can also be very straight forward to just pull in everything at once, which is probably easier for a header only library.

I would be happy to help with the creation of the pack (which should hopefully be able to be integrated into your CI system)

jwellbelove, 2022-10-31:
That sounds like a really good idea.
I'm not going to be able to check that suggestion out at the moment as I'm going to be busy until after the 20th November.
If you want to make a start at looking at what is required, that would be great.

jwellbelove, 2022-11-01:
I've used uVision in the past and I have a copy installed here at home.
I sent an email to Open-cmsis at Linaro to request access to add the ETL to their database.

jwellbelove, 2022-11-09:
I got a reply from Linaro

_Bill Fletcher commented:

Hi John,

Thanks for your interest in the Open-CMSIS-Pack project. The project is focussed on packaging infrastructure and tools rather than building a repository of software packages. Most project participants are involved in IDE development. The concept of a software layer has been mentioned in the aims but is not actively being developed within the collaborative project.

We are always interested to find ways to celebrate great open source projects like ETL but we unfortunately don't currently have a place for it within the Open-CMSIS-Pack project.

Best regards

Bill Fletcher
Linaro LITE Chair_

closed 2022-11-09 by jwellbelove

Jonny-vb, 2022-11-10:
Honestly slightly disappointed by Linaro about this. Their own documentation talks about providing libraries (see below), as well as many already existing in the main pack repository, such as lwip, unity testing, cJSON... I could go on... All of these seem to me to be in a similar boat to ETL.

[image]

But if they don't want it, so be it I guess...