ETLCPP/etl #632: Signed/unsigned warnings

Rasmus-Fink, 2022-11-08:
Hi,
We would like to compile our code with  "-Wsign-conversion" but enabling and using etl::string<> this emits a lot of warnings like 
`C:/projects/integrator/3rdParty/etl/include/etl/basic_string.h:1064:44: warning: conversion to 'etl::string_base::size_type' {aka 'unsigned int'} from 'std::iterator_traits<char*>::difference_type' {aka 'int'} may change the sign of the result [-Wsign-conversion]

 1064 |       const size_type start = etl::distance(begin(), position);
      |                               ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~`

Is there a workaround for this or is this a bug in ETL?
BR

jwellbelove, 2022-11-08:
I'll add `-Wsign-conversion` to my CMake scripts to see if I can eliminate the warnings.
I went through the code a while ago clearing out warnings but I didn't seem t get any from basic_string with `-Wall` & `-Wextra`.

jwellbelove, 2022-11-08:
I've made a start, but it looks like it'll take a few days to do.
Unfortunately I'm away in Malta for two weeks, so I'll not get much time during that period to work on it.
In the meantime you could disable the warning around code that produces it.

mcr-ksh, 2022-11-28:
I haven't gone into details here what this is about, but signed/unsigned conversion errors can lead to serious security vulnerabilities and should not be discarded that easily with a flag

jwellbelove, 2022-11-28:
As a general rule you are correct, but the conversions in the ETL that the compiler is complaining about are values that are well within the ranges of both the signed and unsigned types that are being used. The conversation are only there because the ETL uses a similar API to the STL in which sizes are represented by unsigned types and some calculations assume signed. I am working to make the ETL warning free, but the solution in most cases will merely be the adding of `static_cast` in the appropriate places.

jwellbelove, 2023-05-28:
This branch is still 'in progress'.