ETLCPP/etl #633: Question: Is there a better way to redefine etl string size after interacting with its buffer directly?

basprins, 2022-11-12:
Hi, I am using etl in my microcontroller project and use (maybe abuse?) the string containers to format strings. For example to format a mac address into an etl string. I do this by grabbing the underlying char buffer directly.

```
    void mac_address_to_string(uint64_t mac_address, etl_string_40 &mac_address_string)
    {
        uint8_t *mac_address_bytes = reinterpret_cast<uint8_t*>(&mac_address);

        string::format(
            mac_address_string.data(), 
            mac_address_string.max_size(),
            "%02X%02X%02X%02X%02X%02X",
            mac_address_bytes[5],
            mac_address_bytes[4],
            mac_address_bytes[3],
            mac_address_bytes[2],
            mac_address_bytes[1],
            mac_address_bytes[0]);

        volatile const size_t s1 = mac_address_string.size(); // zero, obviously, etl string couldn't know the size unless it would reevaluate the length of the string with strlen)
        mac_address_string.uninitialized_resize(strlen((const char *)mac_address_string.data()));
        volatile const size_t s3 = mac_address_string.size(); // this works, but feels like a hack
    }
```

The first thing I tried, is to call `repair()` after interacting with the underlying char buffer directly. Then I learned that `repair()` only sets the underlying pointer. Reading the docs:
"This function must be called if the string has been copied via a low level method such as memcpy."

Wouldn't it make sense to also reevaluate the current_size member? Or is there any other way you would suggest?

jwellbelove, 2022-11-13:
You could maybe try using something based on `etl::string_stream`

[https://www.etlcpp.com/string_stream.html](https://www.etlcpp.com/string_stream.html)

```
etl::string<20>    text;
etl::string_stream stream(text);

stream << etl::hex 
       << etl::setw(2) 
       << etl::setfill('0') 
       << mac_address_bytes[5]
       << ":"
       << mac_address_bytes[4]
       << ":"
       mac_address_bytes[3]
       << ":"
       mac_address_bytes[2]
       << ":"
       mac_address_bytes[1]
       << ":"
       mac_address_bytes[0];

const etl::istring& result = stream.str();
```

jwellbelove, 2022-11-13:
_Wouldn't it make sense to also reevaluate the current_size member?_

No, as the destination string after memcpy must be identical to the source, even if the user has inserted nulls in the middle of the string.

jwellbelove, 2022-11-13:
There is also `etl::to_string`

[https://www.etlcpp.com/to_string.html](https://www.etlcpp.com/to_string.html)

basprins, 2022-11-13:
Hi @jwellbelove 
Thanks a lot for your suggestions. Here some results you may or not care about :).

1. The sort of hack with `uninitialized_resize` is my baseline measurement. It works.
2. Using a string_stream for me is not an option. It adds a stack usage of 32 words (word=uint32_t). Every place I would need to format something adds an additional 32. 
3. Changing etl's string::repair with a strlen adds nothing in stack usage, and also works. Of course it will add something, but nothing I can measure, so it's not pushing the high water mark further up. 
4. to_string doesn't work. It clears the underlying buffer, and in my case these will always be the same. I am using std c function interacting with the underlying buffer directly, again to save bytes.

> No, as the destination string after memcpy must be identical to the source, even if the user has inserted nulls in the middle of the string.

Right, I get that in case of `memcpy`, But from where I am standing a small change in the `repair` function would allow user code to go all out on tampering with the underlying buffer and fix it with a simple call to 

Maybe it's not all that interesting for the majority of your users. But since etl should be a good fit for micro controller projects too, maybe its something worth considering? Or make it optional with a #if defined ETL_STRING_REPAIR_FIX_LENGTH or something.

Again, thanks for the help, my problem is solved !

closed 2022-11-13 by basprins

jwellbelove, 2022-11-13:
`etl::to_string` has an optional `append` parameter, which stops the string from being cleared on each call.

I think a separate member function to `repair` would be better to reset the length after manual manipulation.

jwellbelove, 2022-11-13:
Doesn't the C function return the number of characters written to the string?
Can't you use that for the `uninitialized_resize()` instead of `strlen()`?

basprins, 2022-11-13:
That's what I had initially. But I wasn't sure if `unitialized_resize()` is intended for this purpose and might break in future releases of etl e.g. I guess I was overthinking it then :). I'll revert back to that approach.

jwellbelove, 2022-11-13:
That's the scenario `uninitialized_resize()` was added for. For adjusting the length of the string after external manipulation of the string buffer. Also, if you know how long the string is going to be, you can resize without having to do a redundant fill of the buffer.