ETLCPP/etl #634: Strange behavior in move assignment of optional

Chiraffollo, 2022-11-15:
Dear John,

I've found a strange behavior using the `optional` class.
If I use the move assignment operator of the optional, the resulting object has always `valid` flag set and the `storage` is moved.
Was that intended?
Here is a small example:

```
#include <iostream>
#include "etl/optional.h"

using MyOptional = etl::optional<std::uint8_t>;

void print(const char* const label, const MyOptional& opt) {
    std::cout << label << ": " << (opt.has_value() ? "true" : "false") << std::endl;
}

MyOptional get_my_optional() {
    MyOptional result = 8;
    result.reset();
    print("before return", result);
    return result;
}

int main(int argc, char** argv) {
    MyOptional opt;
    opt = get_my_optional();
    print("after return", opt);

    return 0;
}
```

The program results in the output:
```
before return: false
after return: true
```

I expected the move assignment to move both `valid` and `storage`.

Chiraffollo, 2022-11-16:
Should I fix it and make a pull request?

jwellbelove, 2022-11-16:
Fixed 20.35.2

closed 2022-11-16 by jwellbelove