ETLCPP/etl #640: `state_chart` does not support const-qualified actions and guards

Gb268, 2022-11-29:
Current behaviour: `state_chart` does not support const-qualified actions and guards. Minimal example:

```
#include "etl/state_chart.h"
using namespace etl;

struct SM : public state_chart<SM> {
    SM() : state_chart<SM>(*this, transitionTable.begin(), transitionTable.end(), stateTable.begin(), stateTable.end(), 0) {};

    bool always() { return true; };
    // instead, this does not compile:
    // bool always() const { return true; };

    void action() { };
    // instead, this does not compile:
    // void action() const { };

    static const array<SM::state, 1> stateTable;
    static const array<SM::transition, 1> transitionTable;
};

constexpr array<SM::state, 1> SM::stateTable = { SM::state(0, &SM::action, nullptr) };
constexpr array<SM::transition, 1> SM::transitionTable = { SM::transition(0, 0, 0, &SM::action, &SM::always) };
```

jwellbelove, 2022-11-30:
There are difficulties in integrating const-qualified member functions into `etl::start_chart`.
If the class signatures were to be changed to allow const-qualified member functions, then very member function would have to be const-qualified. The only other way to do it would be to add extra const-qualified member function pointers to the state and transition template classes, thereby increasing their size.
The only easy way to make this work would be to `reinterpret_cast` the member function pointers in the initialisation arrays.

jwellbelove, 2022-11-30:
The transition and state classes _could_ hold an indexed union of both function pointer types, but that would also mean that the arrays would have to store indexed unions of both function pointer types too. This would probably be the only efficient way to do it.

jwellbelove, 2022-12-10:
I have a working solution but it requires that the `action`, `guard`, `on_entry` & `on_exit` functions have an extra layer of indirection to allow selection of the correct member function type.
So, more flexibility, but a small overhead.

jwellbelove, 2022-12-29:
I don't really like the overheads that adding support for const-qualified member functions requires, so I'm going to put this on the 'Won't Do' pile.

closed 2022-12-29 by jwellbelove