ETLCPP/etl #641: Wrong usage of enable_if for none-C++11 compilers

Un-done, 2022-12-07:
I stumbled on the following code:
```c++
template <typename U, size_t N>
    ETL_CONSTEXPR array_view(const etl::array<U, N>& a, 
           typename etl::enable_if<etl::is_same<typename etl::remove_cv<T>::type, typename etl::remove_cv<U>::type>::value, void>::type)
    ETL_NOEXCEPT
```
The problem is `enable_if<P,void>`. If the predicate is true the function signature resolves to `array_view(const etl::array<U, N>& a, void)` which is not legal and therefore discarded as well.
Valid solutions could be:
```c++
template <typename U, size_t N>
    ETL_CONSTEXPR array_view(const etl::array<U, N>& a, 
           typename etl::enable_if<etl::is_same<typename etl::remove_cv<T>::type, typename etl::remove_cv<U>::type>::value, void>::type* = 0)  // or "bool" + "= false"
    ETL_NOEXCEPT
```
or 
using the "return"-variation of SFINAE, which works at least for my compiler (IARAVR)

I noticed this issue with `array_view` as well as `span`.

dhebbeker, 2022-12-07:
[Here](https://godbolt.org/z/PoK1sYebf) is an example usage with the compilation error.

jwellbelove, 2022-12-08:
Fixed 20.35.5