ETLCPP/etl #644: `ETL_STRING_TRUNCATION_IS_ERROR` doesn't exist

aunsbjerg, 2022-12-19:
`ETL_STRING_TRUNCATION_IS_ERROR` is mentioned in the [docs](https://www.etlcpp.com/macros.html) but doesn't seem to appear in the codebase since 740e490aedd39b2c066f30b5d54f5f321513b082.

I'm not sure if there is a replacement setting or if it has been deprecated altogether.

jwellbelove, 2022-12-19:
It looks like I changed it to `ETL_ENABLE_ERROR_ON_STRING_TRUNCATION` from 20.26.0.
`ETL_MESSAGES_ARE_VIRTUAL` is no more as messages are always virtual now.
`ETL_USE_MEM_BUILTINS` has also gone as an experimental feature that didn't work very well.
I'll update the documentation.

jwellbelove, 2022-12-19:
And I'll update the CMake files.

closed 2022-12-19 by jwellbelove

rovo89, 2025-05-04:
@jwellbelove https://www.etlcpp.com/string.html still mentions only `ETL_STRING_TRUNCATION_IS_ERROR`.

rovo89, 2025-05-12:
Still so, I think you have just replace one out of three occurences. ;)