ETLCPP/etl #649: test_intrusive_list.cpp fails to compile (-Werror=array-bounds) due to subscript -1 is out of bounds

janhenke, 2022-12-25:
```
[133/288] /usr/bin/x86_64-pc-linux-gnu-g++ -DETL_DEBUG -I/var/tmp/portage/dev-embedded/etlcpp-20.35.8/work/etl-20.35.8/test/../include -isystem /var/tmp/portage/dev-embedded/etlcpp-20.35.8/work/etl-20.35.8/test/UnitTest++/..  -march=native -O2 -pipe -fsanitize=address,undefined -Wall -Wextra -Werror -std=gnu++17 -MD -MT test/CMakeFiles/etl_tests.dir/test_intrusive_list.cpp.o -MF test/CMakeFiles/etl_tests.dir/test_intrusive_list.cpp.o.d -o test/CMakeFiles/etl_tests.dir/test_intrusive_list.cpp.o -c /var/tmp/portage/dev-embedded/etlcpp-20.35.8/work/etl-20.35.8/test/test_intrusive_list.cpp
FAILED: test/CMakeFiles/etl_tests.dir/test_intrusive_list.cpp.o 
/usr/bin/x86_64-pc-linux-gnu-g++ -DETL_DEBUG -I/var/tmp/portage/dev-embedded/etlcpp-20.35.8/work/etl-20.35.8/test/../include -isystem /var/tmp/portage/dev-embedded/etlcpp-20.35.8/work/etl-20.35.8/test/UnitTest++/..  -march=native -O2 -pipe -fsanitize=address,undefined -Wall -Wextra -Werror -std=gnu++17 -MD -MT test/CMakeFiles/etl_tests.dir/test_intrusive_list.cpp.o -MF test/CMakeFiles/etl_tests.dir/test_intrusive_list.cpp.o.d -o test/CMakeFiles/etl_tests.dir/test_intrusive_list.cpp.o -c /var/tmp/portage/dev-embedded/etlcpp-20.35.8/work/etl-20.35.8/test/test_intrusive_list.cpp
In file included from /var/tmp/portage/dev-embedded/etlcpp-20.35.8/work/etl-20.35.8/test/test_intrusive_list.cpp:33:
/var/tmp/portage/dev-embedded/etlcpp-20.35.8/work/etl-20.35.8/test/../include/etl/intrusive_list.h: In function ‘void UnitTest::ExecuteTest(T&, const UnitTest::TestDetails&, bool) [with T = {anonymous}::Suitetest_intrusive_list::SetupFixturetest_sort_compareHelper]’:
/var/tmp/portage/dev-embedded/etlcpp-20.35.8/work/etl-20.35.8/test/../include/etl/intrusive_list.h:617:68: error: array subscript -1 is outside array bounds of ‘{anonymous}::DataNDC1 [1]’ {aka ‘etl::intrusive_list<{anonymous}::ItemNDCNode, etl::bidirectional_link<1> > [1]’} [-Werror=array-bounds]
  617 |       return iterator(static_cast<value_type&>(this->terminal_link));
      |                                                                    ^
/var/tmp/portage/dev-embedded/etlcpp-20.35.8/work/etl-20.35.8/test/test_intrusive_list.cpp:819:16: note: while referencing ‘data1’
  819 |       DataNDC1 data1(unsorted_data.begin(), unsorted_data.end());
      |                ^~~~~
```
While packaging this for Gentoo, I enabled the test suite. Unfortunately the tests do not compile with the above error. Can someone with more insight into the test suite take a look at this? Compiler is `gcc (Gentoo 11.3.1_p20221209 p3) 11.3.1 20221209`.

jwellbelove, 2022-12-28:
I think it may be due to that fact that iterators in `intrusive_list` are constructed from references to nodes and `end()` finds a reference to the `terminal_node`. `terminal_node` does not point to a valid element and, although it is not referenced, it may be upsetting the sanitizer.

Changing the iterator constructor to take a pointer rather than a reference is simple.
I've attached an updated file for you to try.
[intrusive_list.h.txt](https://github.com/ETLCPP/etl/files/10314186/intrusive_list.h.txt)

jwellbelove, 2023-01-20:
Were you able to try the fix?

janhenke, 2023-01-26:
Unfortunately these changes did not fix the issue. Build log with the above file: https://gitlab.com/-/snippets/2489763/raw

jwellbelove, 2023-02-24:
My experiments so far show that the issue is down to the value_type multiple inheriting from two link_type base classes. The error occurs when the value_type is up-cast from one of the base types.

jwellbelove, 2023-02-24:
Currently, the test node is define as follows.
```
typedef etl::bidirectional_link<0> FirstLink;
typedef etl::bidirectional_link<1> SecondLink;

class ItemDCNode : public FirstLink, public SecondLink
```
If the tests are modified so that it is defined as below, then the error disappears.
`class ItemDCNode : public FirstLink`

jwellbelove, 2023-02-25:
I've found the issue.
To resolve it I had to change the code to work with `link_type` pointers and only cast the `value_type` pointers when I needed to access the actual value.
The underlying issue was that `terminal_link`, which is a `link_type` pointer, was being cast to `value_type` pointer.

jwellbelove, 2023-03-08:
Fixed in 20.35.13

closed 2023-03-08 by jwellbelove