ETLCPP/etl #651: Add result<T, void> specialization

aBraM-aBraM, 2022-12-26:
Some errors don't have additional data (think of is_odd(int num) kind of function).

Add a specialization for void error type with default ctor creating an error

jwellbelove, 2022-12-27:
If the error type is `void`, what to do about the member functions

```
const TError& error() const
TError&& error()
```

jwellbelove, 2022-12-27:
Or do you expect the `error()` functions to be not implemented, like `value()` is not implemented for `etl::result<void, TError>`

aBraM-aBraM, 2022-12-27:
I believe it should not be implemented just as result<void, T> doesn't
implement value().

It should also be noted and documented and the user has the responsibility
to use is_value / is_error appropriately.

I feel good about it as all errors related to this are compile-time.

jwellbelove, 2023-01-07:
Fixed 20.35.9

closed 2023-01-07 by jwellbelove