ETLCPP/etl #652: etlcpp-20.35.8 fails to compile: message_packet.h:260:24: error: (etl::imessage)((char)&packet + offsetof(etl::Packet, etl::message_packet<<unnamed>::Message1, <unnamed>::Message2, <unnamed>::Message3>::data.etl::aligned_storage<48, 8

janhenke, 2022-12-28:
Our CI found a bug/compile error (due to -Werror) on GCC 12

```
FAILED: test/CMakeFiles/etl_tests.dir/test_intrusive_list.cpp.o 
FAILED: test/CMakeFiles/etl_tests.dir/test_message_packet.cpp.o 
FAILED: test/CMakeFiles/etl_tests.dir/test_optional.cpp.o 
/var/tmp/portage/dev-embedded/etlcpp-20.35.8/work/etl-20.35.8/test/../include/etl/message_packet.h:260:24: error: ‘*(etl::imessage*)((char*)&packet + offsetof(etl::Packet, etl::message_packet<<unnamed>::Message1, <unnamed>::Message2, <unnamed>::Message3>::data.etl::aligned_storage<48, 8>::type::data[0])).etl::imessage::_vptr.imessage’ may be used uninitialized [-Werror=maybe-uninitialized]
```

The complete build log is attached to the downstream bug report. I would appreciate if someone could look into this issue.

Downstram bug: https://bugs.gentoo.org/888695

jwellbelove, 2022-12-28:
This warning has turned up a few times because the underlying aligned storage is an uninitialised array of char. It is pointless to initialise the array in most circumstances as it will be used to placement new an object into it. Zeroing the array would serve no useful purpose and add additional initialisation overhead.
My solution so far has been to disable the warning around the lines and functions that the compiler has problems with.
I have just upgraded my WSL to GCC-12, so I'll see if I get get the warnings.

jwellbelove, 2022-12-28:
I am using these options in the CMake file
```
if ((CMAKE_CXX_COMPILER_ID MATCHES "GNU") OR (CMAKE_CXX_COMPILER_ID MATCHES "Clang"))
	target_compile_options(etl_tests
			PRIVATE
			-fsanitize=address,undefined
			-Wall
			-Wextra
			-Werror
			)
	target_link_options(etl_tests
			PRIVATE
			-fsanitize=address,undefined
			)
endif ()
```

jwellbelove, 2022-12-28:
Unfortunately all of the test compiled and run without warnings or errors.

Ubuntu 22.04
GCC 12.1.0

jwellbelove, 2022-12-29:
Any ideas on what is different between our two setups?

jwellbelove, 2023-01-20:
Do you still have this issue?

janhenke, 2023-01-26:
https://gitlab.com/-/snippets/2489760/raw

janhenke, 2023-01-26:
Specifically GCC 12.2.1 is now the default compiler. 
```                                                                                                                
gcc (Gentoo 12.2.1_p20230121-r1 p10) 12.2.1 20230121
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
With this version I can test it more easily. And I can reproduce this issue, full build log is in the above snippet.

jwellbelove, 2023-01-26:
I'm off on holiday to Spain tomorrow and I'll just have my laptop with me, but I'll see what I can do to fix this while I'm away.

jwellbelove, 2023-02-28:
I've been trying to get rid of `-Wmaybe-unititialized` warnings, currently in `etl::optional`.
I think that GCC's implementation of the warning is not very good and makes **_many_** false positives.

For example, the code below will generate `-Wmaybe-unititialized` warnings.
But, if you comment out just one of the `etl::optional<S>` declarations then the (false positive) warning goes away!

Internally, the string is part of a union. When default constructed, the union is initialised with the dummy member, but GCC is complaining that the string member is not initialized, which, while true, is irrelevant for a default constructed `etl::optional`.

```
struct S
{
  std::string str;
};

etl::optional<S> data1;
etl::optional<S> data2;
```