ETLCPP/etl #653: Warning when used with CMake

flaviut, 2022-12-30:
When I use this as a dependency in my CMake project (with CPM/FetchContent), I get the following warning: "Version string HEAD-HASH-NOTFOUND retrieved with git describe is invalid".

This is because etl [tries to get the version number with Git first, before falling back to `version.txt`](https://github.com/ETLCPP/etl/blob/master/CMakeLists.txt#L9-L12).

Would it be reasonable to wrap [this in a `PROJECT_IS_TOP_LEVEL` flag](https://cmake.org/cmake/help/latest/variable/PROJECT_IS_TOP_LEVEL.html)? If ETL is being built standalone or on older versions of CMake, first try and get the version from git. If ETL is being build as a subdirectory, first try and get the version `version.txt`. In both cases, fall back to the other on failure.

jwellbelove, 2022-12-31:
That sounds like a reasonable change. Are you able to do the change and create a pull request for this?

jwellbelove, 2023-01-07:
Fixed 20.35.9

closed 2023-01-07 by jwellbelove