ETLCPP/etl #654: Use of the base i* classes (e.g. iqueue_mpmc_mutex)

computerquip, 2022-12-31:
<details>
  <summary>Long-winded rant about what I'm trying to do</summary>
I'm doing a little experiment to see if I can make a totally allocator-free and standard-library-free server library over various platforms where the overlying user of the library must provide buffers to the library. It's not exactly been an easy one with C++ but I was hoping I could use this library to ease some of the pain.<br />
<br />
One of the platforms I started implementing for is Windows. To implement an efficiently scaling server, I'm using IOCP with a threadpool to handle completion packets. To fulfill the requirement of no allocations within the library, the API requires pre-allocating everything the server needs before it starts, including message buffers and per-io-operation data. How the message buffers are implemented is a large contiguous section of memory (required to be aligned to a page). In order for this to work, the accepting thread must allocate a buffer to pass to something like WSASend. When the worker thread receives a completion packet, it either re-uses the buffer or allocates more buffers to send to one or more peers. It's not known which thread does this. The only thing I need is a synchronization mechanism and something to manage the buffers. For this, I wanted to use ETLCPP's iqueue_mpmc_mutex. Upon trying to use it, I realized that it's actually protected so I'm wondering what the correct thing to do here is.
</details>

I have a need to use `queue_mpmc_mutex` with a size not known at compile-time. I realized that there's `iqueue_mpmc_mutex`, however, it can't be used directly. Is it reasonable to derive from these myself and use them as is? Is there a reason why the `i*` classes are all protected?

jwellbelove, 2022-12-31:
The `i` classes are the 'interface/implementation' classes. Their construction is protected as they get the buffer/pool and its size from the derived template class constructor.

Some of the containers have `_ext` variants (`etl::vector_ext` etc) in which the buffer and size information is supplied to the constructor, allowing runtime configuration.

I have a long term plan to give most, if not all, ETL containers a `_ext` variant.

As `queue_mpmc_mutex` does not currently have such a variant you will need to create a class derived from `iqueue_mpmc_mutex` with that functionality.

**A simplified example**
```
template <typename T, const size_t MEMORY_MODEL = etl::memory_model::MEMORY_MODEL_LARGE>
class queue_mpmc_mutex_ext : public etl::iqueue_mpmc_mutex<T, MEMORY_MODEL>
{
private:
  typedef etl::iqueue_mpmc_mutex<T, MEMORY_MODEL> base_t;

public:
  //*************************************************************************
  /// Default constructor.
  //*************************************************************************
  queue_mpmc_mutex_ext(void* buffer, size_t max_size)
    : base_t(reinterpret_cast<T*>(buffer), max_size)
  {
  }
};
```

computerquip, 2022-12-31:
Okay, will do.

closed 2022-12-31 by computerquip