ETLCPP/etl #661: binary dump (to SD card, should make no difference): confirmation this works as I expect, could it be added to documentation somehow?

jerabaul29, 2023-01-17:
Let us say I have a etl vector:

```cpp
etl::vector<uint16_t, MAX_NBR_SAMPLES> samples;
```

and I have a ```size_t write(const void * source, size_t size)``` that writes down to some place (in my case, an SD card) size bytes starting from source (the ```size_t``` return is just number of bytes written).

Is it correct that by doing:

```cpp
write((void *)(&samples), sizeof(samples));
```

I will "simply" be dumping the full vector, including its metadata and its content?

I guess the metadata will be "simply" the current number of elements within, and the content will be a ```uint16_t[]```, right? Is there a way I can know how these will be laid out / is this something I can generally rely on later on when reading / parsing this written dump even if the layout is not part of the public API? If so, it would be very convenient and useful, and would be great to document :) .

I have had a look at the documentation and I could not find information about specifically this, but maybe I missed something.

If course, I could do a loop, and print element by element, but it is quite a bit more work, size may vary depending on the number of elements actually present etc, so doing a "fixed size dump" has its advantages in my opinion :) .

jwellbelove, 2023-01-17:
The vector has contiguous storage so you should be able to dump the raw element data by copying from the pointer returned by the `data()` member function.

`write((void *)(samples.data()), sizeof(uint16_t) * samples.size());`

If you want to dump _all_ of the vector, including the unused storage, then do this...

`write((void *)(samples.data()), sizeof(uint16_t) * samples.max_size());`

You would have to store the size of the vector as well.

jwellbelove, 2023-01-17:
It maybe worthwhile adding a member function to vector to return the size of the type.

```
size_t type_size() const
{
  return sizeof(value_type);
}
```

jerabaul29, 2023-01-17:
I will have to test, but I would guess that ```sizeof(samples)``` would be the size of the whole object (data + information about how far it is filled?) and that dumping the whole object would be fine? But thank you for your advise, I will test a bit and if there are issues with what I was thinking dump directly the .data() thing :) .

jwellbelove, 2023-01-17:
The danger of storing the whole vector object as binary dump is that your code may break if the internal layout of the vector class changes at some point in the future (or if the compiler decides to pack the data in a different way). It would be better to create a set of serialisation functions for each object that you want to store, using the public APIs of the objects.

jerabaul29, 2023-01-17:
Ok, so "my" method should work, but does not rely on public API but on implementation design, so it is not guaranteed to be working in a compatible way with future releases is how I understand it :) . Many thanks for the explanations :) . Given that this is for a really small test / pilot project, that I will freeze all libraries versions and even once I am happy with the firmware compile it, dump the binary and in the future just upload the same binary on the MCU, that is acceptable to me :) .

I think we had already discussed serialization functions some time ago when discussing backup to FRAM or SD card to allow persistence in case of reboot / power loss / critical crash and watchdog reboot. Does that mean that some standardized API for serialization / deserialization associated with the container types could be useful? :)

jwellbelove, 2023-01-20:
I remember talking about that. I started an experiment with a persistence class some time ago, but I haven't looked at it for a while. I should probably give it another go soon.

closed 2023-01-20 by jwellbelove