ETLCPP/etl #673: Adding a ```type_size``` constexpr method (and possibly a "macro like" version too, though not sure if possible) to containers?

jerabaul29, 2023-03-08:
I implemented something over the lines of what is discussed here: https://github.com/ETLCPP/etl/issues/661 for a project.

As you mention, having a function:

```
size_t type_size() const
{
  return sizeof(value_type);
}
```

would make the kind of implementations described there for dumping the content of an array to raw storage a lot easier to "automate" - by now I have to go by hand and tune the types depending on what kind of type I have for my containers elements, this is error prone and brittle as I have to remember to update the code at several locations if I do an update to the types of the elements in the containers. Would it be possible to implement such a function on all containers (and by the way, could it be not just ```const```, but ```constexpr```, so it can be used at compile time if there is need to allocate some raw memory based on it etc? :) ).

---

Also, not sure if it is possible, but is there any way to implement something that works "like a macro" and returns the type itself for coding things more easily regarding "type information propagation"? I would guess this is not possible in C++, but a "class macro" like this would be great if it would be actually possible:

```
my_container.MACRO_ELEM_TYPE;  // this would get replaced by the compiler before compilation by the type of the type of the container elements
```

jwellbelove, 2023-05-02:
I've decided to add static member functions + addition ones for map containers.
I am prefixing them with 'size_of' to tie in with the ETL's etl::size_of() API.
There will also be global functions that do the same on the supplied object.

```
//*************************************************************************
/// Returns the size of the value_type.
//*************************************************************************
static ETL_CONSTEXPR size_t size_of_value_type()
{
  return sizeof(value_type);
}

//*************************************************************************
/// Returns the size of the key_type.
//*************************************************************************
static ETL_CONSTEXPR size_t size_of_key_type()
{
  return sizeof(key_type);
}

//*************************************************************************
/// Returns the size of the mapped_type.
//*************************************************************************
static ETL_CONSTEXPR size_t size_of_mapped_type()
{
  return sizeof(mapped_type);
}
```

jwellbelove, 2023-05-05:
After much experimentation, I've decided that having class member functions is not really the way to go as _every_ container or object must have the required member function(s) added. Global templated functions work better, but they can be more verbose and specialised than the simple code that they implement.

Macros can work a _little_ better in that they don't restrict you to any particular type.

If you want macros then you could try these, but I don't think that they add much to the readability of the code.
```
/// Returns the size of 'Type' defined in the declared type of 'Object'.
#define SIZE_OF_OBJECT_TYPE(Object, Type) sizeof(decltype(Object)::Type) 

i.e.
std::vector<int> v;

// Using macro
constexpr size_t s1 = SIZE_OF_OBJECT_TYPE(v, value_type);

// Not using macro
constexpr size_t s2 = sizeof(decltype(v)::value_type);

// Using member function
constexpr size_t s3 = v.size_of_value_type();

// Using global template function
constexpr size_t s4 = etl::size_of_value_type(v);
``` 
```
/// Returns the size of 'Type' defined in 'Class'.
#define ETL_SIZE_OF_CLASS_TYPE(Class, Type) sizeof(Class::Type)

i.e.
using Vector = std::vector<int>;

// Using macro
constexpr size_t s1 = SIZE_OF_CLASS_TYPE(Vector, value_type);

// Not using macro
constexpr size_t s2 = sizeof(Vector::value_type);

// Using member function
constexpr size_t s3 = Vector::size_of_value_type();

// Using global template function
constexpr size_t s4 = etl::size_of_value_type<Vector>();
```

closed 2023-05-28 by jwellbelove