ETLCPP/etl #676: Getting ETL working in a new VisualGDB project

magene, 2023-03-11:
I hope you don't mind what is probably a dumb question from a relatively inexperienced programmer.  I'm trying to move an embedded C++ CMake project that uses ETL into a VisualGDB project using Visual Studio 2022.  I'm using C++17.  I've gotten most of the problems beaten down except I'm getting 7776 errors that all look like they are related to ETL like these

[image]

I'm guessing that I don't have something setup correctly, probably in the profile.h file.  I copied the cpp17.h file from the etl/profiles folder and renamed it etl_profile.h.  Can you suggest what I'm doing wrong?

Thanks

jwellbelove, 2023-03-11:
Do you have file names and line numbers for some of the errors?

jwellbelove, 2023-03-12:
What tool chain are you using?

jwellbelove, 2023-03-13:
I have downloaded VisualGDB and created a project based on the GCC Arm eabi toolchain.
I need to know what headers and ETL templates you are trying to use.

magene, 2023-03-20:
Hi - I've been out of touch for a while but I'm back on this project starting next week.  I'll get it going again and hopefully have answers for you in the next  couple of days.

thanks for working on this.

jwellbelove, 2023-05-02:
Is this issue still open?

magene, 2023-05-02:
My apologies, I was able to resolve this issue.  Thanks