ETLCPP/etl #677: Change unit test framework to Google Test

jwellbelove, 2023-03-15:
The current UnitTest++ framework no longer appears to be maintained and the ETL's fork has been customised to update it to be compatible with current C++ versions.
I propose updating the unit test framework to the more mainstream Google Test.

jwellbelove, 2023-03-15:
It seems that there is not a current test adaptor for VS2022, which is a deal breaker for me.

jwellbelove, 2023-03-15:
Unfortunately there does not seem to be any proper integration with VS2022 build output windows, making it a bit of a non-starter for me.

jwellbelove, 2023-03-15:
OK, it seems that to get the output to appear in the IDE's output window you must redirect the unit test output to `std::err`

i.e.
`etl.exe 1>&2`

drewr95, 2023-03-16:
> OK, it seems that to get the output to appear in the IDE's output window you must redirect the unit test output to `std::err`
> 
> i.e. `etl.exe 1>&2`

This is what I found as well.  We use googletest to test our embedded architecture.  Using VSCode on Windows, I could only print with std::cerr in my googletests; I haven't tried on linux yet if that's the same case.  So I don't think it's just a VS2022.  Think it's more of a googletest thing?

jwellbelove, 2023-03-16:
> Think it's more of a googletest thing?

Yes. I saw an old issue raised on the googletest GitHub page that asked for the output to go to `std::cerr` instead of `std::cout` (for the reasons we have discovered), but the change was rejected and the redirect was suggested instead.

closed 2025-02-23 by jwellbelove

tzijnge, 2026-02-27:
> Unfortunately there does not seem to be any proper integration with VS2022 build output windows, making it a bit of a non-starter for me.

@jwellbelove and @drewr95 Can you give an example of the things you want printed to the console in one of the tests? I have had good experiences in the past with googletest and Visual Studio. It just prints all the basic info in nice colors. You can stream custom error messages to the EXPECT macros and they also show up in the normal output only if a test fails:

``` cpp
EXPECT_EQ(0, 1) << "0 is not equal to 1";
```

I am currently using it with cmake and VSCode. With the right extensions it's very easy to run/debug individual test. It also integrates with ctest. I think it would be very helpful for ETL as well