ETLCPP/etl #678: Test suite fails on recent Debian system

dkogan, 2023-03-16:
Hi. I'm preparing an etlcpp package for Debian. The test suite fails on the latest tag. Here's the relevant log:

```
dima@fatty:~/projects/etlcpp$ uname -a

Linux fatty 5.19.0-2-amd64 #1 SMP PREEMPT_DYNAMIC Debian 5.19.11-1 (2022-09-24) x86_64 GNU/Linux

dima@fatty:~/projects/etlcpp$ g++ --version

g++ (Debian 12.2.0-14) 12.2.0

dima@fatty:~/projects/etlcpp$ git reset --hard 20.35.14

dima@fatty:~/projects/etlcpp$ mkdir build

dima@fatty:~/projects/etlcpp$ cd build

dima@fatty:~/projects/etlcpp/build$ cmake -DBUILD_TESTS=ON ..

dima@fatty:~/projects/etlcpp/build$ make -j18
[  0%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/AssertException.cpp.o
[  0%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/Test.cpp.o
...
[ 99%] Building CXX object test/CMakeFiles/etl_tests.dir/test_xor_checksum.cpp.o
[ 99%] Building CXX object test/CMakeFiles/etl_tests.dir/test_xor_rotate_checksum.cpp.o
[100%] Linking CXX executable etl_tests
[100%] Built target etl_tests

dima@fatty:~/projects/etlcpp/build$ cd test 

dima@fatty:~/projects/etlcpp/build/test$ ./etl_tests 

==2019137==ERROR: AddressSanitizer failed to allocate 0xdfff0001000 (15392894357504) bytes at address 2008fff7000 (errno: 12)
==2019137==ReserveShadowMemoryRange failed while trying to map 0xdfff0001000 bytes. Perhaps you're using ulimit -v
zsh: IOT instruction  ./etl_tests
```

The machine is a very vanilla Debian, at more-or-less the final configuration that we're going to release soon as Debian/bookworm. amd64 architecture. Anything else you need?

Thanks

jwellbelove, 2023-03-18:
15392894357504 bytes (15393 gigabytes) seems an awful lot of memory to be allocating!
To be honest, I don't run the unit tests from the CMakefiles.txt in the ETL's root directory (that CMake file was written by someone other than me).
Could you try running the unit tests from the `test` directory?

jwellbelove, 2023-03-18:
I've just tried the root `CMakeLists.txt` with the commands you used and it compiled without error under Ubuntu 22.04

dkogan, 2023-03-19:
Hi. Thanks for replying. I dug into this a bit just now, and apparently the issue is with the address sanitizer: it allocates ludicrous amounts of memory, which only works if the kernel allows overcommitting. I did not (`/proc/sys/vm/overcommit_memory` was 2), so asan fails immediately. The etlcpp tests currently build with asan, so they were affected. Can the etlcpp tests be built without any sanitizers? I don't think they should care about my kernel settings: it's just user-space C++ headers.

If cmake is not the preferred way to build and run the tests, what is the preferred way?

Thanks

jwellbelove, 2023-03-19:
If you don't want the address sanitizer to run then remove them from the `CMakeLists.txt` in `/etl/test` (lines 354 & 362)

I use the `CMakeLists.txt` in `/etl/test` to run the unit tests under Ubuntu 22.04 (WSL2) on my Windows machine and for the GitHub CI actions.

dkogan, 2023-03-19:
Hi. I certainly can and will disable the sanitizers for the Debian package. But don't you want to do that in the main tree? Otherwise people's system configs can break your tests.

closed 2023-05-02 by jwellbelove