ETLCPP/etl #680: Placement new does not compile with Zephyr

gabbla, 2023-03-30:
Hello all,

I'd like to use ETL on a Zephyr based project, but I'm having issue using containers.
The following snippet compiles just fine on my pc

```cpp
void foo() {
    etl::vector<int, 10> v;
    v.push_back(1);
}
```

but fails with the following error when compiling with zephyr's g++ (`arm-zephyr-eabi-g++ (Zephyr SDK 0.16.0) 12.2.0`):

```
etl/include/etl/memory.h: In instantiation of 'void etl::create_copy_at(T*, T&&) [with T = int]':
etl/include/etl/vector.h:1064:26:   required from 'void etl::ivector<T>::create_back(rvalue_reference) [with T = int; rvalue_reference = int&&]'
etl/include/etl/vector.h:451:18:   required from 'void etl::ivector<T>::push_back(rvalue_reference) [with T = int; rvalue_reference = int&&]'
src/main.cpp:90:20:   required from here
etl/include/etl/memory.h:1820:5: error: no matching function for call to 'operator new(sizetype, int*&)'
 1820 |     ::new (p) T(etl::move(value));
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
<built-in>: note: candidate: 'void* operator new(unsigned int)'
<built-in>: note:   candidate expects 1 argument, 2 provided
<built-in>: note: candidate: 'void* operator new(unsigned int, std::align_val_t)'
<built-in>: note:   no known conversion for argument 2 from 'int*' to 'std::align_val_t'
```

The build command is the following (redacted):

```
/opt/zephyr-sdk-0.16.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-g++ -DETL_NO_STL -DHSE_VALUE=8000000 -DKERNEL -DSTM32F070xB -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D__PROGRAM_START -D__ZEPHYR__=1 -Isrc -Isrc/include -I/home/gabbla/zephyrproject/zephyr/include -Ibuild/zephyr/include/generated -I/home/gabbla/zephyrproject/zephyr/soc/arm/st_stm32/stm32f0 -I/home/gabbla/zephyrproject/zephyr/lib/posix/getopt/. -I/home/gabbla/zephyrproject/zephyr/drivers -I/home/gabbla/zephyrproject/zephyr/soc/arm/st_stm32/common -I/home/gabbla/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/gabbla/zephyrproject/modules/hal/stm32/stm32cube/stm32f0xx/soc -I/home/gabbla/zephyrproject/modules/hal/stm32/stm32cube/stm32f0xx/drivers/include -I/home/gabbla/zephyrproject/modules/hal/stm32/stm32cube/stm32f0xx/drivers/include/Legacy -I/home/gabbla/zephyrproject/modules/hal/stm32/stm32cube/common_ll/include -isystem /home/gabbla/zephyrproject/zephyr/lib/libc/minimal/include -isystem /home/gabbla/zephyrproject/zephyr/lib/cpp/minimal/include -isystem /opt/zephyr-sdk-0.16.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/include -isystem /opt/zephyr-sdk-0.16.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/include-fixed -isystem /modules/etl/include -fno-strict-aliasing -Os -fcheck-new -std=c++20 -Wno-register -Wno-volatile -fno-exceptions -fno-rtti -imacros /build/zephyr/include/generated/autoconf.h -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m0 -mthumb -mabi=aapcs --sysroot=/opt/zephyr-sdk-0.16.0/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/gabbla/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map==CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/gabbla/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/gabbla/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -D_POSIX_THREADS -nostdinc -nostdinc++ -MD -MT CMakeFiles/app.dir/src/main.cpp.obj -MF CMakeFiles/app.dir/src/main.cpp.obj.d -o CMakeFiles/app.dir/src/main.cpp.obj -c src/main.cpp
```

I'm pretty sure I'm missing something on the configuration side. Any suggestion?

Best regards,
Gab

jwellbelove, 2023-03-30:
Can you try defining `ETL_USING_STD_NEW` as `0` or `1` in your project to see if either one works?

jwellbelove, 2023-03-30:
If your IDE shows which conditional code is active, can you tell me what code is currently enabled in `placement_new.h`?

gabbla, 2023-03-30:
I didn't know about that macro/file. Forcing `ETL_USING_STD_NEW` to `0` fixed the problem! The deduction included the library `<new>`.

closed 2023-03-30 by gabbla

jwellbelove, 2023-03-30:
I there any macro value in Zephyr based projects (or the particular compiler) that could be used to let platform.h automatically decide whether to use <new> or not?

jwellbelove, 2023-03-30:
Does the Zepher compiler not define a <new> include file? From what I can tell, it is a modified GCC compiler.

jwellbelove, 2023-03-31:
Setting macro `ETL_USING_STD_NEW` to `0` was introduced because some platforms (such as Arduino) did not have a `<new>` include, so a I needed to be able to decide whether to include `<new>` or use the ETL's definition. It seems that the Zephyr's `<new>` include does not have a compatible 'placement new'. That is odd and unexpected; I may have to investigate further.

jwellbelove, 2023-03-31:
Could it be that your compiler has a `<new>` include, but which contains no implementation of 'placement new'?

gabbla, 2023-03-31:
The toolchain has its own `new` file ([new.zip](https://github.com/ETLCPP/etl/files/11120647/new.zip)).

The snippet does not build and fails with the same error as before:

Build with C++ 20 support:

```
src/main.cpp: In function 'int main(int, char**)':
src/main.cpp:43:27: error: no matching function for call to 'operator new(sizetype, char*)'
   43 |   ::new (&buffer) char('A');
      |                           ^
<built-in>: note: candidate: 'void* operator new(unsigned int)'
<built-in>: note:   candidate expects 1 argument, 2 provided
<built-in>: note: candidate: 'void* operator new(unsigned int, std::align_val_t)'
<built-in>: note:   no known conversion for argument 2 from 'char*' to 'std::align_val_t'
```

With C++ 11 support:

```
src/main.cpp: In function 'int main(int, char**)':
src/main.cpp:43:27: error: no matching function for call to 'operator new(sizetype, char*)'
   43 |   ::new (&buffer) char('A');
      |                           ^
<built-in>: note: candidate: 'void* operator new(unsigned int)'
<built-in>: note:   candidate expects 1 argument, 2 provided
```

jwellbelove, 2023-03-31:
There seems to be 'placement new' definitions in the file. The question is, why does it think they are not a match?

```
// Default placement versions of operator new.
_GLIBCXX_NODISCARD inline void* operator new(std::size_t, void* __p) _GLIBCXX_USE_NOEXCEPT
{ return __p; }
_GLIBCXX_NODISCARD inline void* operator new[](std::size_t, void* __p) _GLIBCXX_USE_NOEXCEPT
{ return __p; }
```

Could it be because it is expecting a void* and we are giving it a char*?

Can you try this...

int main()
{
  char buffer;

  ::new (static_cast<void*>(&buffer)) char('A');
}

gabbla, 2023-03-31:
Same result:

C++20

```
src/main.cpp: In function 'int main(int, char**)':
src/main.cpp:43:47: error: no matching function for call to 'operator new(sizetype, void*)'
   43 |   ::new (static_cast<void*>(&buffer)) char('A');
      |                                               ^
<built-in>: note: candidate: 'void* operator new(unsigned int)'
<built-in>: note:   candidate expects 1 argument, 2 provided
<built-in>: note: candidate: 'void* operator new(unsigned int, std::align_val_t)'
<built-in>: note:   no known conversion for argument 2 from 'void*' to 'std::align_val_t'
```

C++11

```
src/main.cpp: In function 'int main(int, char**)':
src/main.cpp:43:47: error: no matching function for call to 'operator new(sizetype, void*)'
   43 |   ::new (static_cast<void*>(&buffer)) char('A');
      |                                               ^
<built-in>: note: candidate: 'void* operator new(unsigned int)'
<built-in>: note:   candidate expects 1 argument, 2 provided
```

jwellbelove, 2023-03-31:
It can't find a match, even though the functions are defined in the header!
That doesn't make sense, unless the compiler is not using that header?

The header says it is a GCC component, and I know that the ETL compiles fine with GCC.

How about...

```
#include <new>

int main()
{
  char buffer;

  ::new (&buffer) char('A');
}
```

jwellbelove, 2023-03-31:
The above compiles fine for GCC.

gabbla, 2023-03-31:
Explicitly including `new` does not change the output. Should I open an issue on zephyr side?

rego21, 2023-04-22:
Hi,

So I'm testing ETL with Zephyr and came across this issue. 
To use the standard new from Zephyr you need to add the following to proj.conf:
CONFIG_LIB_CPLUSPLUS=y
CONFIG_NEWLIB_LIBC=y

This should enable the standard library.