ETLCPP/etl #682: hfsm::start() does not update state for call_on_enter_state = true

jwelz, 2023-04-06:
Prerequisite: My hfsm statemachine looks like
```
state1
   state1_1
      state1_1_1
      state1_1_2
   state1_2
...
```
Calling hfsm::start(true) will call do_enters() for state1, state1_1 and state1_1_1 and return 2 (i.e. index for state1_1_1). But p_state remains state_list[0]. 
Because the enter actions are executed p_state should point to state1_1_1.

The proposed change is

```
void start(bool call_on_enter_state = true) ETL_OVERRIDE
{
......
            etl::fsm_state_id_t next_state = do_enters(ETL_NULLPTR, p_state, true);
            if (ifsm_state::No_State_Change != next_state)
            {
                p_state = state_list[next_state];
            }
}
```

jwellbelove, 2023-04-07:
I don't have a lot of free time at the moment. Would you be able to modify the unit hfsm unit tests to highlight the failure with the current code?

jwelz, 2023-04-07:
It's a honor.

jwellbelove, 2023-04-27:
Have you had a chance to update the tests yet?

jwelz, 2023-04-27:
Sorry - I was quite busy. I'd like to create a merge request + branch in GitHub - do I need additional rights/invitation to the project? I read the documentation but don't see the described buttons etc. (sorry - I'm used to Azure DevOps or GitLab).

jwellbelove, 2023-04-28:
No worries about being busy. I've been off coding for a while as I ended up getting appendicitis about a week and a half ago and was in hospital for a few days.

jwellbelove, 2023-05-01:
I've added a hotfix branch for this issue.
hotfix/#682-hfsm-start()-does-not-update-state-for-call_on_enter_state-=-true

jwellbelove, 2023-05-01:
If I try a modified state machine that returns anything other than `ifsm_state::No_State_Change` from the initial state's `on_enter_state()` then a `etl::fsm_state_composite_state_change_forbidden` exception is raised.

jwellbelove, 2023-05-01:
Actually, I've just thrown together a simple unit test for the feature, as I'm soon to do a new release.

closed 2023-05-01 by jwellbelove

jwelz, 2023-05-01:
Sorry for beeing so lame.

jwellbelove, 2023-05-01:
I sure you're very busy with other far more important things than the ETL. :-)