ETLCPP/etl #685: Redeclaration of C++ built-in type ‘char8_t’ error when using C++20

Tagussan, 2023-04-11:
Hi. I'm using g++ 9.4.0.
If I compile my project using etl using `-std=c++17`, everything compiles fine. However, when I compile it with `-std=c++2a`, I get the following compile error:
```
In file included from etl/string.h:34,
                 from UART_EVM.hpp:5,  <- my file
....
                 from main.cpp:5,  <- my file
etl/platform.h:325:20: error: redeclaration of C++ built-in type ‘char8_t’ [-fpermissive]
```

Is this library compatible with C++20?

jwellbelove, 2023-04-11:
`char8_t` is not defined by the ETL if it determines that you are using C++20 (`__cplusplus >= 202002L`)

-std=c++2a gives varying values of __cplusplus for different compilers.
It looks like GCC defines `__cplusplus` as `201709L` in GCC (8.x -> 10.x) so you could try this in `etl_profile.h`.

`#define ETL_CPP20_SUPPORTED (__cplusplus >= 201709L)`

That should compile OK as the ETL uses very little of C++20 at the moment.

Tagussan, 2023-04-13:
Thank you! This error message is solved by this fix.
However, then I faced another error:
```
In file included from etl/basic_string.h:46,
                 from etl/string.h:35,
....

etl/memory.h: In function ‘constexpr T* etl::construct_at(T*, TArgs&& ...)’:
etl/memory.h:966:17: error: ‘construct_at’ is not a member of ‘std’; did you mean ‘etl::construct_at’?
  966 |     return std::construct_at(p, etl::forward<TArgs>(args)...);
      |                 ^~~~~~~~~~~~
etl/memory.h:964:22: note: ‘etl::construct_at’ declared here
  964 |   ETL_CONSTEXPR20 T* construct_at(T* p, TArgs&&... args)
      |
```
I'm using tag version 20.35.12.

jwellbelove, 2023-04-13:
As g++ 9.4.0 has only partial support for C++20 then it may be better to tell the ETL not to use any C++20 at all.

Define `ETL_CPP20_SUPPORTED` as `0`

Tagussan, 2023-04-14:
Sorry for bothering you because of my side's problem. Everything is okay now, so I'm closing this.
Thank you for your support!

closed 2023-04-14 by Tagussan