ETLCPP/etl #696: fsm/hfsm self-transitions

eloisesko, 2023-05-15:
Given that either `No_State_Change` or a state ID can be returned by `on_event` function , I would expect a different behavior between returning :
-  `No_State_Change` : internal transition
-  the current state id : exit and re-enter the state (self-transition)

Currently both perform an internal transition.
Looking at the code, it may be the intended behaviour but it would be useful to manage both cases.

jwellbelove, 2023-05-15:
In the older FSM, returning the current state was intended to indicate that you did not want to change state. The HFSM was built on the FSM, but returning the current state was illegal as a return for on_enter_state() in the HFSM. So, for the HFSM `ifsm_state::No_State_Change` was added. If returning the current state id and `ifsm_state::No_State_Change` did different things then it would break backward compatibility, unless a compile or runtime option was added to enable this.

jwellbelove, 2023-05-15:
Maybe add a enable_self_transitions() member function?

eloisesko, 2023-05-31:
Sorry for the late answer. What about adding a new constant like `ifsm_state::Self_Transition` to be returned by `on_event` to handle self transitions instead ?

jwellbelove, 2023-08-13:
Will be added in the next release.

jwellbelove, 2023-08-29:
Added 20.38.0

closed 2023-08-29 by jwellbelove