ETLCPP/etl #697: Documentation: shared-pool list

ryanpowers, 2023-05-18:
I noticed an issue with the documentation for shared-pool lists. Not sure if the API used to be different, but the documentation page doesn't match the latest implementation: [https://www.etlcpp.com/list.html](https://www.etlcpp.com/list.html)

This applies to the `Shared Pools` section at the top and `Shared pool example` section at the bottom. 

On the documentation page, it shows using a list of type `etl::list<T, 0>` and instantiating a pool of type `etl::pool<list::pool_type, N>`, but the list type should be `etl::list_ext<T>`. This error is in both sections. I've included the example section below (with the fix). The top section has a lot more explanation and I don't feel comfortable proposing modifications for it, as it includes some other use cases that I haven't made use of.

Current Documentation:
```cpp
// The list type
typedef etl::list<Point, 0> List;

// The shared pool. Maximum of 10 items.
etl::pool<List::pool_type, 10> pool;

// The vector of lists with shared pools.
etl::vector<List, 3> vector_of_lists(3, List(pool));
```
Corrected:

```cpp
// The list type
typedef etl::list_ext<Point> List;

// The shared pool. Maximum of 10 items.
etl::pool<List::pool_type, 10> pool;

// The vector of lists with shared pools.
etl::vector<List, 3> vector_of_lists(3, List(pool));
```

jwellbelove, 2023-05-18:
Thanks, I'll update that.

jwellbelove, 2023-05-28:
Fixed

closed 2023-05-28 by jwellbelove