ETLCPP/etl #698: Behaviour of bidirectional_link nodes

godavogel, 2023-05-24:
The behaviour of intrusive list nodes (bidirectional_link) seems to me quite confusing and can possibly be improved. When an node is linked up within an intrusive list and the node's "unlink" method is called, the node indeed is taken from the list, but the node's method "is_linked" still returns "true". The same applies to the containers pop_front() and alike methods.

This is because "unlink" only modifies "etl_previous->etl_next" and "etl_next->etl_previous" but leaves "etl_previous" and "etl_next" unchanged, while "is_linked" only checks for "etl_next" or "etl_previous" being unequal NULL. An additional call to the "clear" method is required to also clear "etl_previous" and "etl_next".

Wouldn't it be an improvement to also reset "etl_previous" and "etl_next" to NULL when "unlink" is called? 

The ETL version I refer to is 20.35.14.

jwellbelove, 2023-05-25:
Thanks, I'll fix that, unless you want to do a pull request with the changes.
If you do, don't forget to first update the intrusive link tests with to demonstrate failure with the issue you've highlighted.

godavogel, 2023-05-25:
Great. If that is helpful for you, I would be happy to contribute the required changes as you suggested. But since I've never done that on github before I'm unsure about to doing. Do I need to create a fork or can I simply create a pull request "in place" by committing/pushing into a branch? To be honest, I've already created a fork, but I'm in doubt if this is the right way....

On the other hand, it would be absolutely fine for me, if you do the change by yourself. ;-)

jwellbelove, 2023-05-31:
I planning to do a release today, so I'll update the code.

jwellbelove, 2023-06-02:
Fixed 20.36.0

closed 2023-06-02 by jwellbelove

godavogel, 2023-06-06:
Thanks a lot, it works perfectly well. I was on holiday the last week, so I couldn't look after that.

I'm still wondering, if in case of the API function "unlink" a termination of the endpoints is also required when a range of elements are unlinked. Isn't it possible to split a linked list (maybe several times) in a way, so that a single element remains? 

Example of change:

```C++
  unlink(TLink& first, TLink& last)
  {
    if (&first == &last)
    {
      first.unlink();
    }
    else
    {
      if (last.etl_next != ETL_NULLPTR)
      {
        last.etl_next->etl_previous = first.etl_previous;
      }

      if (first.etl_previous != ETL_NULLPTR)
      {
        first.etl_previous->etl_next = last.etl_next;
      }

      // terminate endpoints
      first.etl_previous = ETL_NULLPTR;
      last.etl_next      = ETL_NULLPTR;
    }
  }
```