ETLCPP/etl #700: Compilation error for [] map operator

r2r0, 2023-06-02:
The problem appeared after switching to the newest and dreatest ETL version.
I ponder if expression 
```c++
::new (etl::addressof(node.value.first))  key_type(key);
```
in the map.h at line 975 is correct.
`addressof` returns `const void*` but placement `::new` requires `void*`.

The simple map element assignement `interfaceGroupMap[fieldName] = value;` generates (sometimes) compilation error.
Where key in interfaceGroupMap is a simple scoped enum and element type is `uint8_t`.
```
ccache /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-g++ -DCORE_CM4 -DFILE_ID_APP_CONFIG=aurora_file_id.h -DGSL_NO_IOSTREAMS -DHSE_VALUE=32000000 -DKERNEL -DLFS_CONFIG=zephyr_lfs_config.h -DMBEDTLS_CONFIG_FILE=\"config-tls-generic.h\" -DNDEBUG -DNXPBUILD__PHHAL_HW_RC663 -DCONFIG_BOARD_HID_AURORA -DPH_OSAL_ZEPHYR -DPICOLIBC_INTEGER_PRINTF_SCANF -DSTM32WB55xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -DZCBOR_CANONICAL -DZCBOR_STOP_ON_ERROR -D_FORTIFY_SOURCE=1 -D_POSIX_C_SOURCE=200809 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -DMAGIC_ENUM_CONFIG_FILE=\"magic_enum_config.hpp\" -
...
fdiagnostics-show-option -DINLINED -fno-strict-aliasing -Os -fcheck-new -std=c++17 -Wno-register -fno-exceptions -fno-rtti -imacros /home/hid/aurora/app/cmbuild/Release/zephyr/include/generated/autoconf.h -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee --sysroot=/opt/zephyr-sdk/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/hid/aurora/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/hid/aurora/app=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/hid/aurora/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/hid/aurora=WEST_TOPDIR -ffunction-sections -fdata-sections --specs=picolibc.specs -MD -MT CMakeFiles/app.dir/src/iso7816/atr.cpp.obj -MF CMakeFiles/app.dir/src/iso7816/atr.cpp.obj.d -o CMakeFiles/app.dir/src/iso7816/atr.cpp.obj -c /home/hid/aurora/app/src/iso7816/atr.cpp
In file included from /home/hid/aurora/app/src/iso7816/atr.hpp:22,
                 from /home/hid/aurora/app/src/iso7816/atr.cpp:16:
/home/hid/aurora/app/../extern/etl/include/etl/map.h: In instantiation of 'etl::imap<TKey, TMapped, TKeyCompare>::mapped_type& etl::imap<TKey, TMapped, TKeyCompare>::operator[](const_key_reference) [with TKey = ISO7816::Atr::AtrInterfaceBytesFieldChar; TMapped = std::optional<unsigned char>; TKeyCompare = etl::less<ISO7816::Atr::AtrInterfaceBytesFieldChar>; mapped_reference = std::optional<unsigned char>&; const_key_reference = const ISO7816::Atr::AtrInterfaceBytesFieldChar&]':
/home/hid/aurora/app/src/iso7816/atr.cpp:369:32:   required from here
/home/hid/aurora/app/../extern/etl/include/etl/map.h:975:30: error: invalid conversion from 'const void*' to 'void*' [-fpermissive]
  975 |         ::new (etl::addressof(node.value.first))  key_type(key);
      |                ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~
      |                              |
      |                              const void*
In file included from /opt/zephyr-sdk/arm-zephyr-eabi/arm-zephyr-eabi/include/c++/12.2.0/bits/new_allocator.h:34,
                 from /opt/zephyr-sdk/arm-zephyr-eabi/arm-zephyr-eabi/include/c++/12.2.0/arm-zephyr-eabi/thumb/v7e-m/nofp/bits/c++allocator.h:33,
                 from /opt/zephyr-sdk/arm-zephyr-eabi/arm-zephyr-eabi/include/c++/12.2.0/bits/allocator.h:46,
                 from /opt/zephyr-sdk/arm-zephyr-eabi/arm-zephyr-eabi/include/c++/12.2.0/memory:64,
                 from /home/hid/aurora/app/../extern/etl/include/etl/private/addressof.h:37,
                 from /home/hid/aurora/app/../extern/etl/include/etl/iterator.h:37,
                 from /home/hid/aurora/app/../extern/etl/include/etl/algorithm.h:43,
                 from /home/hid/aurora/app/../extern/etl/include/etl/array.h:35,
                 from /home/hid/aurora/app/src/utils/bytearray.hpp:16,
                 from /home/hid/aurora/app/src/utils/simple_serialization.hpp:22,
                 from /home/hid/aurora/app/src/iso7816/atr.hpp:17:
/opt/zephyr-sdk/arm-zephyr-eabi/arm-zephyr-eabi/include/c++/12.2.0/new:174:65: note:   initializing argument 2 of 'void* operator new(std::size_t, void*)'
  174 | _GLIBCXX_NODISCARD inline void* operator new(std::size_t, void* __p) _GLIBCXX_USE_NOEXCEPT
      |                                                           ~~~~~~^~~
```

The problem appeared after switching to the newest and greatest ETL version.

Unfortunately I cannot reproduce problem in simple example and I got error not for all uses of etl::map only in my code.

ETL version is 20.36.0.

jwellbelove, 2023-06-02:
The lines at 975 & 976 are missing a cast.
```
  ::new ((void*)etl::addressof(node.value.first))  key_type(etl::move(key));
  ::new ((void*)etl::addressof(node.value.second)) mapped_type();
```

florianhumblot, 2023-06-02:
according to https://stackoverflow.com/a/69304324/4461980 it would be undefined behavior to cast away the const-ness of the `node.value.first`, wouldn't it be better (and safer) to not introduce the `const` qualifier on `value_type`?

jwellbelove, 2023-06-02:
`Data_Node& node = create_data_node(); // 'node' has not been constructed yet.`

jwellbelove, 2023-06-03:
The only other way to do this, if your compiler is set to error on const casts, is the have `create_data_node() `return a `char*` and placement-new to the offset addresses of the elements in the `Data_Node` class.

jwellbelove, 2023-06-03:
Scrub the above idea. `offsetof` does not work with non-POD classes.

jwellbelove, 2023-07-08:
Fixed 20.37.0

closed 2023-07-08 by jwellbelove