ETLCPP/etl #701: Trying to use STL type traits even though STL is disabled

chibicitiberiu, 2023-06-02:
I would like to use ETL in a freestanding x86 environment, for OS development. I don't have any sort of libc implementation. I was hoping that ETL, being an embedded oriented library, would work in such an environment.

I am having a strange problem related to type_traits.h. Even though I set the ETL_NO_STL macro, for some reason I'm getting errors like this:

```
etl/include/etl/type_traits.h:810:52: error: expected template-name before '<' token
  810 |   struct integral_constant : std::integral_constant<T, VALUE> {};
      |                                                    ^
etl/include/etl/type_traits.h:810:52: error: expected '{' before '<' token
etl/include/etl/type_traits.h:822:48: error: expected template-name before '<' token
  822 |   struct bool_constant : std::integral_constant<bool, B> { };
      |                                                ^

```

Looking in `type_traits.h`, this is very strange because this should only get compiled when using STL:

```
79:     #if ETL_NOT_USING_STL || ETL_CPP11_NOT_SUPPORTED
...
800:    #else // Condition = ETL_USING_STL && ETL_USING_CPP11
```

I can't figure out what the problem seems to be. I'm thinking that maybe there is some mismatched `#endif` somewhere. Or, maybe I am doing something wrong, but I can't figure out what.

I was able to create a minimal example in which I can reproduce this issue [in this repo](https://github.com/chibicitiberiu/etl-issue-repro). To reproduce the issue, you will need a GCC cross compiler targeting the i686-elf platform. The provided `setup_toolchain.sh` automates the process of downloading and building the compiler, just install the [build dependencies](https://wiki.osdev.org/GCC_Cross-Compiler#Installing_Dependencies) and then run `./setup_toolchain.sh .toolchain`. Alternatively, there might be packages available in the repos of some Linux distributions. After that, just run `make`.

jwellbelove, 2023-06-02:
Thanks, I'll take a look at that asap.

jwellbelove, 2023-06-03:
How/where are you setting the `ETL_NO_STL` macro?

chibicitiberiu, 2023-06-03:
Thank you for your help. I set it in `etl_profile.h`, and I also tried using `-DETL_NO_STL` in the makefile.

jwellbelove, 2023-06-03:
Are you able to see in your editor what lines are enabled in platform.h?
Specifically

```
#if defined(ETL_NO_STL)
  #define ETL_USING_STL     0
  #define ETL_NOT_USING_STL 1
#else
  #define ETL_USING_STL     1
  #define ETL_NOT_USING_STL 0
#endif
```

chibicitiberiu, 2023-06-03:
Unfortunately no, vscode isn't picking up the proper compiler options. I can test by adding `#warning` statements and try to compile, like this:

```
#if ETL_USING_STL
#warning In typetraits.h ETL_USING_STL=1
#else
#warning In typetraits.h ETL_USING_STL=0
#endif

#if ETL_NOT_USING_STL
#warning In typetraits.h ETL_NOT_USING_STL=1
#else
#warning In typetraits.h ETL_NOT_USING_STL=0
#endif
```

I added this in typetraits.h before the #if statement (at line 79), and then inside the #else statement (at line 802), and in platform.h right after the `#if defined(ETL_NO_STL)`. This was the output:

```
In file included from etl/include/etl/limits.h:35,
                 from etl/include/etl/platform.h:35,
                 from etl/include/etl/string_view.h:36,
                 from test.cpp:1:
etl/include/etl/type_traits.h:82:2: warning: #warning In typetraits.h ETL_USING_STL=0 [-Wcpp]
   82 | #warning In typetraits.h ETL_USING_STL=0
      |  ^~~~~~~
etl/include/etl/type_traits.h:88:2: warning: #warning In typetraits.h ETL_NOT_USING_STL=0 [-Wcpp]
   88 | #warning In typetraits.h ETL_NOT_USING_STL=0
      |  ^~~~~~~
etl/include/etl/type_traits.h:817:2: warning: #warning In typetraits.h else condition ETL_USING_STL=0 [-Wcpp]
  817 | #warning In typetraits.h else condition ETL_USING_STL=0
      |  ^~~~~~~
etl/include/etl/type_traits.h:823:2: warning: #warning In typetraits.h else condition ETL_NOT_USING_STL=0 [-Wcpp]
  823 | #warning In typetraits.h else condition ETL_NOT_USING_STL=0
      |  ^~~~~~~
etl/include/etl/platform.h:83:2: warning: #warning In platform.h ETL_USING_STL=0 [-Wcpp]
   83 | #warning In platform.h ETL_USING_STL=0
      |  ^~~~~~~
etl/include/etl/platform.h:87:2: warning: #warning In platform.h ETL_NOT_USING_STL=1 [-Wcpp]
   87 | #warning In platform.h ETL_NOT_USING_STL=1
      |  ^~~~~~~
etl/include/etl/type_traits.h:834:52: error: expected template-name before '<' token
  834 |   struct integral_constant : std::integral_constant<T, VALUE> {};
      |                                                    ^
```

chibicitiberiu, 2023-06-03:
I think I understand what is happening... it's a circular dependency.

In platform.h at line 35 there is `#include <limits.h>`. Since the etl include directory is added as an include path, it's picking up `limits.h` from etl which includes `type_traits.h`, so it ends up in type_traits before the ETL_USING_STL/ETL_NOT_USING_STL macros are actually defined.

jwellbelove, 2023-06-03:
My standard advice is that the path is set to the ETL's 'include' directory so that the ETL headers are included as `#include "etl/string_view.h"` otherwise it can cause other header issues such as `"string.h"`

chibicitiberiu, 2023-06-03:
Yes, that is a great idea. Thank you very much for your help, I think the problem is solved.

closed 2023-06-03 by chibicitiberiu