ETLCPP/etl #702: etl::message not `trivially_copyable`

florianhumblot, 2023-06-05:
Ever since this commit: https://github.com/ETLCPP/etl/commit/f3ace52884b87abb3729dccfbf0dfd74fb62f6d4 the etl::message class (and imessage) are no longer trivially copyable. If you used to not have `ETL_MESSAGES_ARE_VIRTUAL` defined, your messages would be 2 bytes in size and be trivially copyable.
In the latest etl version these messages are now 8 bytes large, and are no longer trivially copyable, which creates a decent overhead when passing a lot of messages around.
Is there a way to alter the design so that it is again possible to satisfy the requirement `std::is_trivially_copyable` ? The size of the messages is secondary, the 6 extra bytes aren't too bad.

florianhumblot, 2023-06-05:
Having the message be trivially copyable is pretty awesome because it also forces the developers that implement new messages to think carefully about what they put in those messages too :) If I can help with implementing this just let me know and I'll try to make some time for it

jwellbelove, 2023-08-13:
The trade-off of getting rid of the virtual function is that the base message type will no longer be an empty class, as `etl::imessage` will have to contain a member storing the message ID.
The ID must be passed down to the base class constructor.

**Virtual**
```
  //***************************************************************************
  // Message interface.
  //***************************************************************************
  class imessage
  {
  public:

    virtual ~imessage()
    {
    }

    ETL_NODISCARD virtual etl::message_id_t get_message_id() const ETL_NOEXCEPT = 0;
  };

  //***************************************************************************
  // Message type.
  //***************************************************************************
  template <etl::message_id_t ID_, typename TParent = etl::imessage>
  class message : public TParent
  {
    ETL_STATIC_ASSERT((etl::is_base_of<etl::imessage, TParent>::value), "TParent is not derived from etl::imessage");

  public:

    static ETL_CONSTANT etl::message_id_t ID = ID_;

    ETL_NODISCARD etl::message_id_t get_message_id() const ETL_NOEXCEPT ETL_OVERRIDE
    {
      return ID;
    }
  };

```

**Non-virtual**
```
  //***************************************************************************
  // Message interface.
  //***************************************************************************
  class imessage
  {
  public:

    ETL_NODISCARD etl::message_id_t get_message_id() const
    {
      return id;
    }

  protected:

    imessage(etl::message_id_t id_)
      : id(id_)
    {
    }

  private:

    const etl::message_id_t id;
  };

  //***************************************************************************
  // Message type.
  //***************************************************************************
  template <etl::message_id_t ID_, typename TParent = etl::imessage>
  class message : public TParent
  {
  public:

    ETL_STATIC_ASSERT((etl::is_base_of<etl::imessage, TParent>::value), "TParent is not derived from etl::imessage");

    message()
      : TParent(ID)
    {
    }

    static ETL_CONSTANT etl::message_id_t ID = ID_;
  };
```

jwellbelove, 2023-08-13:
I done a little experimentation, which has highlighted that `etl::message_packet` cannot be used with a non-virtual `etl::message` as it destroys the messages that it stores through the base class destructor.

Also, having a non-virtual base message class requires the derived classes to implement constructors that take the message ID, which makes using messages a lot more fiddly and error prone.

florianhumblot, 2023-08-14:
@jwellbelove what if you do it like it used to be? Store the message id in the `etl::imessage`, enable the virtual construtor/destructor using a preprocessor macro and have the `etl::imessage::message_id`  memeber be protected or even public (As long as it's const, its visibility shouldn't be an issue). 

That way you can have `etl::message_packet` use the virtual interface, and keep the trivially copyable messages if one doesn't need the message_packets.

jwellbelove, 2023-08-29:
Added 20.38.0
Define `ETL_MESSAGES_ARE_NOT_VIRTUAL` to activate.

closed 2023-08-29 by jwellbelove