ETLCPP/etl #703: How to compose message routers

chenlijun99, 2023-06-05:
Use case:

I have a series of orthogonal state machines, where each one is a message router. All these state machines are executed in an actor task which is itself a queued message router. The actor should accepts all the messages accepted by the orthogonal state machines that are running inside it.

I think it should be possible to merge the message types of each message router into a single parameter packs, similar to https://stackoverflow.com/questions/56591727/stdtuple-cat-but-only-with-unique-elements.

I tried to look in the ETL documentation but couldn't find anything. Is there already some mechanism implemented in ETL?

jwellbelove, 2023-06-05:
Which message router class is given this combined set of message types?
Is the intention to be able to filter the messages in the actor?

chenlijun99, 2023-06-05:
> Which message router class is given this combined set of message types?

The actor (which is a message router) is given this combined set of message types

>  Is the intention to be able to filter the messages in the actor?

For now I don't think so. For how I use the actor, actually the actor doesn't even need to know anything about the messages that are getting sent to it. It simply forwards messages from its message queue to all the orthogonal message routers that are executing within it and that accept that such messages.

Now that I think, I could make all the actors be message buses and then subscribe all the message routers within each actor to the actor message bus. But then all the messages must be sent to all the actors, which will wake up actor threads needlessly.

chenlijun99, 2023-06-05:
> Are the messages not sent to all of the actors anyway. I'm not sure I see the difference in operation between a queued message router actor and a queued message bus actor

Ok. The difference that I was thinking about between message routers and message buses is that a message router carries with its type the set of messages that it accepts. Thus, I thought that making actors be message routers instead of message buses could somehow help me achieve "Are the messages not sent to all of the actors anyway", in other words, message are sent to actors only if at least one of the message routers within it handle such messages. So. I thought that if there is a way to compose message routers, then using the type information of the set of combined messages that the composed actor message routers carry, I can hook the actor message routers only to the producers that produce these messages.

Perhaps I should mention a further complication that I would incur if I go on with this approach: I end up having message routers (in this case actors) for which I don't know at programming time the exact set of messages it takes. I could know by digging into the code of each message routers that are composed into the actor message router, but that is not so maintainable.

Sidenote: My current architecture is that there are classes that are "message producers" and that take in a `etl::imessage_router`. Message producers simply send the messages they produce to the `etl::imessage_router` they are given.

Now, the knowledge of which message router consumes which messages is essential to be able to hook message producers with the message routers. If all the messages produced by a message producer are consumed only by a single message router, then I can pass the consumer message router directly to the message producer. OTOH, if there are multiple consumer message routers who need a certain message from a message producer, then I need to pass to the producer a `etl::message_bus` and then subscribe all the consumers to the message bus.

With the approach of composing message routers, it would be unmaintainable to manually hook the *combined* message routers with the message producers. The goal would be that if at a lower level I change the messages handled by a message router, then I wouldn't need to change the initial "hooking" code.

This is becoming quite complex... But yet I feel it's quite natural to reach this kind of message passing architecture. Am I missing some more idiomatic way to do something similar to this with ETL?