ETLCPP/etl #704: Compilation error with avr-gcc-7.3

jputcu, 2023-06-05:
After updating the code to "etl/20.36.0" I get the following compilation errors:

~~~~
include\etl\parameter_pack.h:135:74: error: template definition of non-template 'constexpr const size_t etl::parameter_pack<TTypes>::index_of_type<T>::value'
   constexpr size_t parameter_pack<TTypes...>::template index_of_type<T>::value;
                                                                          ^~~~~
~~~~

jputcu, 2023-06-05:
Removing the `::template` fixed the error.

jputcu, 2023-06-05:
But isn't the compiler correct, my mingw gcc 13.1 doesn't complain after removing the `template`?

jwellbelove, 2023-06-05:
I'm pretty sure the keyword is necessary according to the standard. The ETL's tests include compilation with GCC 12.

jwellbelove, 2023-06-09:
I'll check in the Git file history to see when it was added.

jwellbelove, 2023-06-11:
Fixed in Pull Request #707

closed 2023-06-11 by jwellbelove