ETLCPP/etl #711: intrusive queue, corruption with doubled push of static objects

SimonFilgis, 2023-06-08:
We are using this queue with 10 different static events.
If one event is pushed twice, the queue is corrupted.
We would like to add:
Explicit Null of object after pop so we can see if a object is linked or not from higher levels (is_linked()).
Assert if a already linked object is pushed.
Do you have any thoughts about this?

jwellbelove, 2023-06-09:
I'll take a look at that.
This is similar to the earlier #698 issue.

jwellbelove, 2023-06-11:
"Assert if a already linked object is pushed."

Other alternatives could be relinking the item to the end of the queue or ignoring items that are already linked.

SimonFilgis, 2023-06-11:
"Other alternatives could be relinking the item to the end of the queue or ignoring items that are already linked."

Another approch could be "merge". Leave position but update data. 

Adding priorities is another thing.

We will have discussions tomorrow. I'm glad to see, that you are onboard! Thank you!!!

jwellbelove, 2023-06-11:
"Another approach could be "merge". Leave position but update data."
The data doesn't need to be updated, as the item in the queue is the same object that was pushed.

SimonFilgis, 2023-06-11:
Right.

The is_linked() check before push is all the "merge" strategy would need.

SimonFilgis, 2023-06-12:
> Other alternatives could be relinking the item to the end of the queue or ignoring items that are already linked.

The downside would be, that the queue has some "inner methodology" that should be located more on system-design level.

jwellbelove, 2023-06-12:
I was working on the principle that "ignoring items that are already linked" would at least be _defined behaviour_, rather than _undefined behaviour_ (messing up the container state)
The `push` could return a bool status to indicate success/failure.

On the other hand, requiring the user to check `is_linked` before calling `push` is also a valid option, though it opens the code to issues if the user forgets to do so when a double push is likely to occur.

Raising an `ETL_ASSERT` is also possible, possibly combined with "ignoring items that are already linked".

I've opened a discussion on [Slack](https://etlcpp.slack.com/archives/C7SJ45VFB/p1686504689985399) to get other opinions on this.

jwellbelove, 2023-06-20:
There is a small issue to get round with the forward intrusive links and containers in that. at the moment, the first item pushed will still appear to be unlinked as the last item in the queue/list is marked by a nullptr 'next' pointer. For the items to appear as linked they will need to be linked to a terminator item. I will need to look at how this would work in conjunction with the ETL link utility functions.

jwellbelove, 2023-07-08:
Fixed 20.37.0

closed 2023-07-08 by jwellbelove