ETLCPP/etl #712: Unable to use etl::optional with non-default-constructible POD objects

eloisesko, 2023-06-08:
The following code won't compile with version 20.36.. It did with 20.23.0 when there was no specialization of `optional` for POD types and also works with STL.

```cpp
#include "etl/optional.h"

struct MyPODObject {
  MyPODObject() = delete;
  int value;
};

int main() {
  etl::optional<MyPODObject> optionalObject;

  optionalObject.emplace(MyPODObject{1});

  return 0;
}
```
A workaround to make the code compile is to explicitly define the destructor for `MyPODObject`, which makes it non-POD.

jwellbelove, 2023-06-08:
There is an old issue open about `etl::optional` and deleted copy constructors. This sounds similar.
I'll have a look at this again.

jwellbelove, 2023-06-11:
What C++ standard are you using?

jwellbelove, 2023-06-11:
OK, this appears to fail to compile with C++20 on VS2022.
For C++14, C++17 & C++20, the class is seen as non-POD.

```
    //*************************************************************************
    struct MyPODObject
    {
      MyPODObject() = delete;
      int value;
    };

    TEST(test_optional_pod_emplace_bug_712)
    {
      etl::optional<MyPODObject> optionalObject;

      bool is_pod1 = etl::is_pod<MyPODObject>::value; // false for C++14, C++17 & C++20
      bool is_pod2 = std::is_pod<MyPODObject>::value; // false for C++14, C++17 & C++20

      optionalObject.emplace(MyPODObject{ 1 }); // Compilation error for C++20
    }
```

jwellbelove, 2023-06-11:
If the line `MyPODObject() = delete;` is removed then it is seen as POD.

jwellbelove, 2023-06-11:
If I try to make a variable I get a compilation error.

`MyPODObject mpo{ 1 };`

`Error C2440 'initializing': cannot convert from 'initializer list' to '`anonymous-namespace'::Suitetest_optional::MyPODObject'`

jwellbelove, 2023-06-12:
From the standard.

_A POD struct is a non-union class that is both a trivial class and a standard-layout class

A trivial class is a class that is trivially copyable and has one or more default constructors (12.1), all of which are either trivial or deleted **and at least one of which is not deleted**._

eloisesko, 2023-06-12:
> What C++ standard are you using?

I'm using C++17 with GCC 12.2.1. The result is the same with C++20, the class is considered as POD with GCC.
But from the standard I understand I shouldn't be the case.

jwellbelove, 2023-06-13:
I've tried the code on Windows below using VS2022, GCC 8.1.0 and Clang 15.0.1

VS2022: MyPODObject is POD: false
GCC: MyPODObject is POD: true
CLang: MyPODObject is POD: false

jwellbelove, 2023-06-13:
I suppose the quickest solution would be to disable the specialisation for GCC.

jwellbelove, 2023-06-13:
Or just remove the specialisation!

jwellbelove, 2023-06-13:
I can make it work by re-defining the storage for the POD specialisation.

```
struct storage_type
{
  storage_type()
  {
  }

  union union_type
  {
    union_type()
      : dummy(0)
    {
    }

    char dummy;
    T value;
  };

  union_type u;
};

storage_type storage;
```
This allows the storage to contain a `T` that is not default constructed. The non-POD specialisation uses a similar technique.

eloisesko, 2023-06-13:
Great, looks like a good solution !

jwellbelove, 2023-07-08:
Fixed 20.37.0

closed 2023-07-08 by jwellbelove