ETLCPP/etl #720: etl::optional rvalue value_or() broken

fermentedfly, 2023-06-27:
Hi,

I have an object whose get method returns an optional

```
struct xxx_bug_helper
{
    float value{5.0f};

    etl::optional<float> get() const
    {
        return value;
    }
};
```

If I chain `get()` and `value_or()` I get undefined behaviour

```
xxx_bug_helper helper{};
float value = helper.get().value_or(1.0); // failure, value is neither 5.0 nor 1.0
```

It works as expected if I instead assign the result of `get()` to a variable

```
xxx_bug_helper helper{};
auto temp = helper.get();
float value = temp.value_or(1.0); // value is 5.0
```

Chaining the methods worked fine previously, the new behavior may be related to changes introduced in #690 (i.e. if I delete the rvalue variant of value_or(), all is well)

Regards,
Manuel

jwellbelove, 2023-07-01:
It appears to be an incorrect function signature for `value_or`.
I'm fixing it in the next release, but here's the updated code.

jwellbelove, 2023-07-01:
Dragging a zip file doesn't seem to work, so it's a txt.
[optional.txt](https://github.com/ETLCPP/etl/files/11926709/optional.txt)

fermentedfly, 2023-07-03:
Hi,
Thanks for investigating.

jwellbelove, 2023-07-08:
Fixed 20.37.0

closed 2023-07-08 by jwellbelove