ETLCPP/etl #727: redeclaration of char8_t compilation error when using C++20 with avr-gcc 9.2

jputcu, 2023-07-06:
Using latest development version, sha:309eebe764f8e5aabf39315b47db9bbd3279c2a3, of etl I get the following error when compiling with C++20 on avr-gcc 9.2:
~~~~
avr-g++.exe -DETL_NO_CPP_NAN_SUPPORT -DETL_NO_HUGE_VAL_SUPPORT -DETL_NO_STL -DETL_TARGET_DEVICE_GENERIC -DETL_TARGET_OS_NONE -DF_CPU=14745600 -mmcu=atmega128 -g -funsigned-char -mrelax -ffunction-sections -fdata-sections -mcall-prologues -fno-rtti -fno-exceptions -fno-threadsafe-statics -Os -DNDEBUG -std=c++2a
...
include/etl/platform.h:337:20: error: redeclaration of C++ built-in type 'char8_t' [-fpermissive]
  337 |     typedef int8_t char8_t;
      |                    ^~~~~~~
~~~~~

jputcu, 2023-07-06:
Also a small warning(not really worth another ticket) when adding `-Wpedantic`:
~~~~
include\etl\type_traits.h(1990,4): warning: extra ';' [-Wpedantic]
		 1990 |   };
		      |    ^
~~~~

jwellbelove, 2023-07-06:
OK, I thought you meant an additional issue to the `char8_t`.

The proposed solution was to define the following, either as project settings or in `etl_profile.h`.

`#define ETL_CPP20_SUPPORTED 0`
`#define ETL_NO_SMALL_CHAR_SUPPORT 0`

jputcu, 2023-07-06:
Actually the issue was #704.
I'm using `conan` with all the `ETL` defines in a profile, to keep it clean I will have to start looking into `etl_profile.h` to limit the command line options.
`-DETL_NO_CPP_NAN_SUPPORT -DETL_NO_HUGE_VAL_SUPPORT -DETL_NO_STL -DETL_TARGET_DEVICE_GENERIC -DETL_TARGET_OS_NONE`

closed 2023-07-08 by jwellbelove