ETLCPP/etl #731: Keil Compiler 5 and minmax_push/pop.h

CurtisAMU, 2023-07-11:
I'm setting up my Keil armcc.exe V5.06 update 6 (build 750) to work with ETL (awesome job, btw!) and noticed that min/max macros were causing compile errors, when I include etl/numeric.h. It includes a minmax_pop.h, that "restores" min/max macros, which in my installation of Keil were not defined at all (to my own surprise, actually). Thus, minmax_pop.h restores two macros that did not exist when minmax_push.h was included, leaving the system in an altered state.

Here are my 2¢.

minmax_pop.h:
```
#if !defined(ETL_COMPILER_GREEN_HILLS) && !defined(ETL_COMPILER_IAR) && !defined(ETL_COMPILER_TASKING)
  #if !defined(ETL_COMPILER_ARM5)
    #pragma pop_macro("min")
    #pragma pop_macro("max")
  #else
    #define min(a,b) ((a)<(b)?(a):(b))
    #define max(a,b) ((a)<(b)?(b):(a))
  #endif
#endif

```
These two macros seem not to be defined unconditionally by armcc.
(Also, I do not need to define __STDC_LIMIT_MACROS for armcc to compile successfully)

I gave this a quick try and it seems to work:
minmax_push.h
```
#if !defined(ETL_COMPILER_GREEN_HILLS) && !defined(ETL_COMPILER_IAR) && !defined(ETL_COMPILER_TASKING)
  #if !defined(ETL_COMPILER_ARM5)
    #pragma push_macro("min")
    #pragma push_macro("max")
  #endif

  #ifdef min
    #ifndef RESTORE_MIN
      #define RESTORE_MIN
    #endif
    #undef min
  #endif
  #ifdef max
    #ifndef RESTORE_MAX
      #define RESTORE_MAX
    #endif
    #undef max
  #endif
#endif

```

minmax_pop.h:
```
#if !defined(ETL_COMPILER_GREEN_HILLS) && !defined(ETL_COMPILER_IAR) && !defined(ETL_COMPILER_TASKING)
  #if !defined(ETL_COMPILER_ARM5)
    #pragma pop_macro("min")
    #pragma pop_macro("max")
  #else
    // undef RESTORE_MIN/RESTORE_MAX be considered
    #ifdef RESTORE_MIN
      #define min(a,b) ((a)<(b)?(a):(b))
    #endif
    #ifdef RESTORE_MAX
      #define max(a,b) ((a)<(b)?(b):(a))
    #endif
  #endif
#endif
```

Does that make sense?

jwellbelove, 2023-07-11:
I would add an `ETL_` prefix to `RESTORE_MIN` and `RESTORE_MAX` to minimise any possibility of macro name clashes.

jwellbelove, 2023-07-12:
I've played around with slightly modified versions that detect mismatched calls to the header.

```
#ifndef ETL_PUSHED_MIN_MAX
  #if !defined(ETL_COMPILER_GREEN_HILLS) && !defined(ETL_COMPILER_IAR) && !defined(ETL_COMPILER_TASKING)
    #if !defined(ETL_COMPILER_ARM5)
      #pragma push_macro("min")
      #pragma push_macro("max")
    #endif

    #ifdef min
      #ifndef ETL_RESTORE_MIN
        #define ETL_RESTORE_MIN
      #endif
      #undef min
    #endif

    #ifdef max
      #ifndef ETL_RESTORE_MAX
        #define ETL_RESTORE_MAX
      #endif
      #undef max
    #endif

    #define ETL_PUSHED_MIN_MAX
  #endif
#else
  #error minmax_push without matching pop
#endif
```

```
#ifdef ETL_PUSHED_MIN_MAX
  #if !defined(ETL_COMPILER_GREEN_HILLS) && !defined(ETL_COMPILER_IAR) && !defined(ETL_COMPILER_TASKING)
    #if !defined(ETL_COMPILER_ARM5)
      #pragma pop_macro("min")
      #pragma pop_macro("max")
    #else
      #ifdef ETL_RESTORE_MIN
        #define min(a,b) ((a)<(b)?(a):(b))
        #undef ETL_RESTORE_MIN
      #endif
  
      #ifdef ETL_RESTORE_MAX
        #define max(a,b) ((a)<(b)?(b):(a))
        #undef ETL_RESTORE_MAX
      #endif
    #endif
    
    #undef ETL_PUSHED_MIN_MAX
  #endif
#else
  #error minmax_pop without matching push
#endif
```

CurtisAMU, 2023-07-12:
That's helpful.
I'm still not sure if it is worth handling min and max separately at all. They typically act as a pair, thus, ETL_RESTORE_MIN_MAX rather than ETL_RESTORE_MIN and ETL_RESTORE_MAX would probably be sufficient. Well, it's there now.

jwellbelove, 2023-07-12:
You are probably correct about them always coming as a pair.

jwellbelove, 2023-07-25:
Fixed 20.37.2

closed 2023-07-25 by jwellbelove