ETLCPP/etl #738: etl::expected assignment operator won't work when TValue and TError are the same type

Vitmark97, 2023-08-01:
Recent change in assignment for etl::expected result in issues when TValue and TError are the same type.
For example:
```cpp
etl::expected<int, int> test_exp{1};
test_exp.has_value(); // true

test_exp = etl::unexpected<int>(456);
test_exp.has_value(); // still true
```

---
I tried changing the assignment back from
```cpp
storage = etl::move(ue.error());
```
to
```cpp
storage.template emplace<Error_Type>(etl::move(ue.error()));
```

But this change results in compilation error in test_expected.cpp when move assigning form error. I'm not sure how to solve this.
```
In file included from /etl/test/../include/etl/variant.h:40,
from /etl/test/../include/etl/expected.h:40,
from /etl/test/test_expected.cpp:31:
/etl/test/../include/etl/private/variant_variadic.h: In instantiation of 'typename etl::variant_alternative<Index, etl::variant<VTypes ...> >::type etl::variant<TTypes>::emplace(TArgs&& ...) [with long unsigned int Index = 2; TArgs = {{anonymous}::ErrorM}; TTypes = {etl::monostate, {anonymous}::ValueM, {anonymous}::ErrorM}; typename etl::variant_alternative<Index, etl::variant<VTypes ...> >::type = {anonymous}::ErrorM]':
/etl/test/../include/etl/expected.h:472:43:   required from 'etl::expected<TValue, TError>& etl::expected<TValue, TError>::operator=(etl::expected<TValue, TError>::unexpected_type&&) [with TValue = {anonymous}::ValueM; TError = {anonymous}::ErrorM; etl::expected<TValue, TError>::unexpected_type = etl::unexpected<{anonymous}::ErrorM>]'
/etl/test/test_expected.cpp:394:38:   required from here
/etl/test/../include/etl/private/variant_variadic.h:722:38: error: use of deleted function '{anonymous}::ErrorM::ErrorM(const {anonymous}::ErrorM&)'
722 |       return *static_cast<type*>(data);
|                                      ^
/etl/test/test_expected.cpp:121:5: note: declared here
121 |     ErrorM(const ErrorM&) = delete;
|     ^~~~~~
```

jwellbelove, 2023-08-02:
Until I figure out a solution to this I would suggest one of the following workarounds.

1. Wrap the error type in a class that mimics the error type.
3. Use an `enum class` as the error type.

```
enum class Error : int
{
  Error1,
  Error2
};

etl::expected<int, Error> test_exp{1};
test_exp.has_value(); // Returns true

test_exp = etl::unexpected<Error>(Error::Error1);
test_exp.has_value(); // Returns false
```

jwellbelove, 2023-08-11:
Fixed 20.37.3

closed 2023-08-11 by jwellbelove