ETLCPP/etl #739: Could not compile ETL with C++11

alexletov, 2023-08-02:
Could not compile ETL with gcc 9.4.0 (C++11) with error

```
include/etl/parameter_pack.h:139:74: error: template definition of non-template 'constexpr const size_t etl::parameter_pack<TTypes>::index_of_type<T>::value'
   constexpr size_t parameter_pack<TTypes...>::template index_of_type<T>::value;
                                                                          ^~~~~
```

Environment:
ETL version: 20.37.2
gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.1) 
gcc-linaro-6.3.1-2017.02-x86_64_arm-linux-gnueabihf cross compile toolchain

alexletov, 2023-08-02:
Proposed solution: [etl.739.patch](https://github.com/ETLCPP/etl/files/12241434/etl.739.patch)

jwellbelove, 2023-08-03:
I think that the solution may be a bit more nuanced.
I compile the unit tests under C++11 and GCC-12 does not error.
Could this be a GCC-9 and not a C++11 issue?

jwellbelove, 2023-08-03:
I think the template keyword may not be allowed in this use case in C++11 (although later compilers may be happy with it).

["template definition of non-template" error while using nested templates](https://stackoverflow.com/questions/24547961/template-definition-of-non-template-error-while-using-nested-templates)

alexletov, 2023-08-03:
> I think that the solution may be a bit more nuanced. I compile the unit tests under C++11 and GCC-12 does not error. Could this be a GCC-9 and not a C++11 issue?

Yes, it looks like this issue is only for gcc-9 from linaro package mentioned in the description. Anyway I have build in many different configurations with c++11 and starting from gcc-10 there is no issue with this code, but gcc-9 does not work. probably the proposed [fix ](https://github.com/ETLCPP/etl/issues/739#issuecomment-1662352305) could work for C++11 as this code should work fine in C++14

alexletov, 2023-08-03:
These code also works fine:

```
#if ETL_USING_CPP17
  template <typename T, typename... TTypes>
  inline constexpr size_t parameter_pack_v = etl::parameter_pack<TTypes...>::template index_of_type<T>::value;
#elif ETL_USING_CPP14
  template <typename... TTypes>
  template <typename T>
  constexpr size_t parameter_pack<TTypes...>::template index_of_type<T>::value;
#else
  template <typename... TTypes>
  template <typename T>
  constexpr size_t parameter_pack<TTypes...>::index_of_type<T>::value;
#endif
```

jwellbelove, 2023-08-03:
I think the last commit changed to an incorrect `#if ETL_USING_CPP17/#else/#endif`

I think it should be this.

```
#if ETL_USING_CPP17
  template <typename T, typename... TTypes>
  inline constexpr size_t parameter_pack_v = etl::parameter_pack<TTypes...>::template index_of_type<T>::value;
#endif

#if ETL_USING_CPP14
  //***********************************
  template <typename... TTypes>
  template <typename T>
  constexpr size_t parameter_pack<TTypes...>::template index_of_type<T>::value;
#else
  //***********************************
  template <typename... TTypes>
  template <typename T>
  constexpr size_t parameter_pack<TTypes...>::index_of_type<T>::value;
#endif
```

alexletov, 2023-08-03:
> I think the last commit changed to an incorrect `#if ETL_USING_CPP17/#else/#endif`
>
> I think it should be this.
>
> ```
> #if ETL_USING_CPP17
>   template <typename T, typename... TTypes>
>   inline constexpr [...]

This code works fine as well.

closed 2023-08-11 by jwellbelove