ETLCPP/etl #742: Build errors when compiling with arm-none-eabi-c++ 13 in freestanding

jferch, 2023-08-11:
Hi everyone,

I was just trying our build with the newest gcc compiler version for bare-metal arm and I get the following errors while compiling the latest release of the ETL (rev. f9da1cde9bace4edd9f1ef4edadf44daac9eb877):

```
In file included from /usr/arm-none-eabi/include/c++/13.2.0/cmath:41,
                 from /usr/arm-none-eabi/include/c++/13.2.0/math.h:36,
                 from /workspaces/.../etl/include/etl/profiles/determine_compiler_language_support.h:34,
                 from /workspaces/.../etl/include/etl/platform.h:130,
                 from /workspaces/.../etl/include/etl/crc16_ccitt.h:34,
/usr/arm-none-eabi/include/c++/13.2.0/bits/requires_hosted.h:34:4: error: #error "This header is not available in freestanding mode."
   34 | #  error "This header is not available in freestanding mode."
```

```
% >> arm-none-eabi-c++ --version
arm-none-eabi-c++ (Arch Repository) 13.2.0
```

Building the tests with gcc 13 passed probably since its hosted (thanks for the latest release btw!) but I couldn't find anything useful while researching this, building with the older v12.2 still worked without issues. Any ideas how this can be solved?

jwellbelove, 2023-08-11:
Is the STL enabled for the ETL?
Or has ETL_NO_STL been defined?

jwellbelove, 2023-08-20:
I'm finding it difficult to get a definitive list of what headers are not available in "freestanding mode", but it seems that `<math.h>` is unavailable.

I'm just testing to see where there may be unnecessary includes of this header.

jferch, 2023-08-28:
Yes, this is probably implementation specific. I've found this [overview ](https://en.cppreference.com/w/cpp/freestanding) and tried to find out which headers include the one that provides the error message if that's helpful:

```
[dev@0fdbdf7d5f86 13.2.0]$ grep -rni . -e 'requires_hosted.h'
./ext/pod_char_traits.h:37:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/bitmap_allocator.h:32:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/ropeimpl.h:1231:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/random:34:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/memory:61:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/vstring.h:34:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/codecvt_specializations.h:38:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/functional:61:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/numeric:61:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/cmath:34:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/stdio_filebuf.h:34:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/new_allocator.h:32:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/vstring.tcc:35:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/mt_allocator.h:32:#include <bits/requires_hosted.h> // getenv
./ext/string_conversions.h:34:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/extptr_allocator.h:39:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/random.tcc:35:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/malloc_allocator.h:32:#include <bits/requires_hosted.h> // malloc
./ext/pool_allocator.h:45:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/rb_tree:61:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/throw_allocator.h:49:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/sso_string_base.h:33:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/enc_filebuf.h:32:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/vstring_util.h:35:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/vstring_fwd.h:35:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/stdio_sync_filebuf.h:34:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/algorithm:61:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/debug_allocator.h:45:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/rc_string_base.h:33:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./ext/slist:47:#include <bits/requires_hosted.h> // std::allocator
./ext/rope:48:#include <bits/requires_hosted.h> // GNU extensions are currently omitted
./chrono:35:#include <bits/requires_hosted.h> // for <ctime> and clocks
./vector:60:#include <bits/requires_hosted.h> // container
./iomanip:38:#include <bits/requires_hosted.h> // iostreams
./future:34:#include <bits/requires_hosted.h> // concurrency
./locale:38:#include <bits/requires_hosted.h> // locales
./string:38:#include <bits/requires_hosted.h> // containers
./random:34:#include <bits/requires_hosted.h> // OS-dependent random
./condition_variable:34:#include <bits/requires_hosted.h> // threading primitive
./stop_token:32:#include <bits/requires_hosted.h> // concurrency
./streambuf:38:#include <bits/requires_hosted.h> // iostreams
./cmath:41:#include <bits/requires_hosted.h>
./unordered_map:34:#include <bits/requires_hosted.h> // container
./stack:60:#include <bits/requires_hosted.h> // containers
./format:34:#include <bits/requires_hosted.h> // for std::string
./tr1/cctype:32:#include <bits/requires_hosted.h> // TR1
./tr1/ctime:32:#include <bits/requires_hosted.h> // TR1
./tr1/wctype.h:32:#include <bits/requires_hosted.h> // TR1
./tr1/stdlib.h:32:#include <bits/requires_hosted.h> // TR1
./tr1/cstdint:34:#include <bits/requires_hosted.h> // TR1
./tr1/random:35:#include <bits/requires_hosted.h> // TR1
./tr1/memory:35:#include <bits/requires_hosted.h> // TR1
./tr1/functional:34:#include <bits/requires_hosted.h> // TR1
./tr1/cwctype:34:#include <bits/requires_hosted.h> // TR1
./tr1/cmath:34:#include <bits/requires_hosted.h> // TR1
./tr1/unordered_map:34:#include <bits/requires_hosted.h> // TR1
./tr1/float.h:32:#include <bits/requires_hosted.h> // TR1
./tr1/tgmath.h:32:#include <bits/requires_hosted.h> // TR1
./tr1/cfloat:32:#include <bits/requires_hosted.h> // TR1
./tr1/array:34:#include <bits/requires_hosted.h> // TR1
./tr1/limits.h:32:#include <bits/requires_hosted.h> // TR1
./tr1/utility:34:#include <bits/requires_hosted.h> // TR1
./tr1/type_traits:34:#include <bits/requires_hosted.h> // TR1
./tr1/complex.h:32:#include <bits/requires_hosted.h> // TR1
./tr1/ctgmath:32:#include <bits/requires_hosted.h> // TR1
./tr1/cstdio:34:#include <bits/requires_hosted.h> // TR1
[... 67 lines cut ...]
./charconv:34:#include <bits/requires_hosted.h> // for error codes
./unordered_set:34:#include <bits/requires_hosted.h> // containers
[... cut ...]
./ios:38:#include <bits/requires_hosted.h> // iostreams
./spanstream:34:#include <bits/requires_hosted.h> // iostreams
./istream:38:#include <bits/requires_hosted.h> // iostreams
./list:60:#include <bits/requires_hosted.h> // containers
./bits/requires_hosted.h:23:/** @file bits/requires_hosted.h
./regex:34:#include <bits/requires_hosted.h> // string and container heavy
./queue:60:#include <bits/requires_hosted.h> // containers
./valarray:36:#include <bits/requires_hosted.h> // <cmath> dependant
```

jferch, 2023-11-22:
Our setup is somewhere in between a freestanding and hosted environment, for our use-case simply removing the freestanding compilation flag is fine. Issue can be closed.

closed 2023-11-22 by jferch