ETLCPP/etl #745: Assertion triggered in hash.h when compiling for MSP430

ekoeppen, 2023-08-19:
When compiling with gcc-13 with MSP430 as the target (here using PlatformIO and a custom GCC), the following assertion is triggered:

```
.pio/libdeps/exp_msp430g2/Embedded Template Library/include/etl/hash.h:192:38: error: static assertion failed: size_t smaller than type
  192 |     ETL_STATIC_ASSERT(sizeof(size_t) >= sizeof(wchar_t), "size_t smaller than type");
      |                       ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~

```

`size_t` is indeed 16 bits, whereas `wchar_t` is 32 bits.

ekoeppen, 2023-08-19:
It seems a specialisation along the lines of the one for the other larger types (e.g. `long` or `long long`) fixes this:

```
  //***************************************************************************
  /// Specialisation for wchar_t.
  ///\ingroup hash
  //***************************************************************************
  template<>
  struct hash<wchar_t>
  {
    size_t operator ()(wchar_t v) const
    {
      // If it's the same size as a size_t.
      if ETL_IF_CONSTEXPR(sizeof(size_t) >= sizeof(v))
      {
        return static_cast<size_t>(v);
      }
      else
      {
        uint8_t* p = reinterpret_cast<uint8_t*>(&v);
        return private_hash::generic_hash<size_t>(p, p + sizeof(v));
      }
    }
  };
```

jwellbelove, 2023-08-29:
Fixed 20.38.0

closed 2023-08-29 by jwellbelove