ETLCPP/etl #746: Missing etl::legacy::variant overload of emplace with 0 Arguments

Un-done, 2023-08-22:
It should be possible to use variant::emplace to construct an object without arguments.
This is possible for C++11 but not for C++98.
See [Compiler Explorer](https://godbolt.org/z/YKqn1fTv6)

I think `variant_legacy.h` is just missing 
```c++
      //***************************************************************************
      /// Emplace with no constructor parameters.
      //***************************************************************************
      template <typename T>
      T& emplace()
      {
        ETL_STATIC_ASSERT(Type_Is_Supported<T>::value, "Unsupported type");

        destruct_current();
        ::new (static_cast<T*>(data)) T();
        type_id = Type_Id_Lookup<T>::type_id;

        return *static_cast<T*>(data);
      }
```

If there is no reason for this to be missing, I can make a pull request.

jwellbelove, 2023-08-22:
Actually, non-of the legacy (C++03 compatible) emplace overloads have a zero argument variant.
I'll go through the library and add them. It should be a fairly quick job.

jwellbelove, 2023-08-29:
Added 20.38.0

closed 2023-08-29 by jwellbelove