ETLCPP/etl #747: constexpr ctor for etl::vector

skorokhod, 2023-08-22:
Clang 16 generates error 'error: constexpr function's return type '...' is not a literal type'
while I'm trying to use vector in constexpr function like:

```
struct xxx
{
    tel::vector<int, 10> a;
};

constexpr xxx deserialize()
{
    return xxx{
        .a = etl::vector<int, 10>{},
    };
};
```

jwellbelove, 2023-08-23:
Most containers in the ETL are not `constexpr` as they use 'placement new' to construct elements into the container's storage. 'placement new' cannot be used in a `constexpr` function.
In C++20, functions such as `std::construct_at` were introduced, which _are_ `constexpr` and can be used to create `constexpr` containers. These functions have been given special status by the compiler to be able to construct to a memory location (using placement new) and be `constexpr`. The C++ committee decided that no user defined function should also have this ability, so making third party libraries unable to define `constexpr` containers unless the STL is used.

The only way the ETL could have `constexpr` containers is to either create specialisations for types that are trivially constructible etc or only enable them when the ETL is used with C++20 & STL.

jwellbelove, 2023-08-23:
https://www.appsloveworld.com/cplus/100/159/placement-new-and-delete-in-constexpr-functions

closed 2023-09-28 by jwellbelove