ETLCPP/etl #749: Code change causes warning with armcc compiler

CurtisAMU, 2023-08-28:
https://github.com/ETLCPP/etl/blob/b38afd7fc98f50a00b207b184506049f05ab7120/include/etl/private/vector_base.h#L171

The extra semicolon causes a warning if `ETL_DEBUG_COUNT` is not defined:
```
..\..\..\Lib\ETL\include\etl/private/vector_base.h(171): warning:  #381-D: extra ";" ignored
      ETL_DECLARE_DEBUG_COUNT;   ///< Internal debugging.
```

We have a zero warning policy for our project, which treats this as an error. It would be okay for debugging to have a warning to ignore but preferably not on release builds.

jwellbelove, 2023-08-28:
Does it warn with `void;`

jwellbelove, 2023-08-28:
Forget that last suggestion

jwellbelove, 2023-08-28:
What about `void(0);`

CurtisAMU, 2023-08-28:
`ETL_DECLARE_DEBUG_COUNT;` is a declaration and `(void)0;` does not work there; it's an expression.
The former
`ETL_DECLARE_DEBUG_COUNT   ///< Internal debugging.`
without the semicolon worked. What was the problem with that? It looks odd without the semicolon but should work.

This also works with armcc:
`#define ETL_DECLARE_DEBUG_COUNT struct SDebugCountDummy {}`

CurtisAMU, 2023-08-28:
A dummy typedef would also do.

jwellbelove, 2023-08-28:
There will also be this warning if `ETL_NO_CHECKS` is defined for `ETL_ASSERT` etc.

jwellbelove, 2023-08-28:
Do the other macros cause a warning?
`ETL_INCREMENT_DEBUG_COUNT` etc

CurtisAMU, 2023-08-28:
Actually no. Empty statements are okay, it seems.

CurtisAMU, 2023-08-28:
> Do the other macros cause a warning? `ETL_INCREMENT_DEBUG_COUNT` etc

Not as far as I can see.

CurtisAMU, 2023-08-28:
They are not used in a class declaration.

jwellbelove, 2023-08-28:
If I try an empty statement in a Keil project using ARM compiler V6 I get

`warning: empty expression statement has no effect; remove unnecessary ';' to silence this warning [-Wextra-semi-stmt]`

CurtisAMU, 2023-08-28:
One of those complementary differences between compiler V5 and V6. There `(void)0;` would help in both cases, I'd expect.

jwellbelove, 2023-08-28:
Ignore that file, there's an error.

CurtisAMU, 2023-08-29:
Looks good, the warning is gone.

jwellbelove, 2023-09-04:
Fixed 20.38.1

closed 2023-09-04 by jwellbelove