ETLCPP/etl #754: ambiguous `etl::bit_stream_reader::bit_stream_reader()` for some `etl::span`

dhebbeker, 2023-09-01:
Some template specializations of `etl::span` can not be passed as argument to `etl::bit_stream_reader::bit_stream_reader()`.

When passing as argument to the constructor:

- ✔️ OK: `etl::span<uint8_t>`
- ❌ Error: `etl::span<const uint8_t>`
- ❌ Error: `etl::span<uint8_t, num_bytes>`
- ❌ Error: `etl::span<const uint8_t, num_bytes>`

```plain
<source>:19:82: error: call of overloaded 'bit_stream_reader(const etl::span<const unsigned char>&, etl::endian::enum_type)' is ambiguous
   19 |     ETL_STATIC_CONSTANT etl::bit_stream_reader bits(bit_span, etl::endian::little);
      |                                                                                  ^
In file included from <source>:1:
/opt/compiler-explorer/libs/etl/include/etl/bit_stream.h:1045:5: note: candidate: 'etl::bit_stream_reader::bit_stream_reader(etl::span<unsigned char>, etl::endian)'
 1045 |     bit_stream_reader(etl::span<unsigned char> span_, etl::endian stream_endianness_)
      |     ^~~~~~~~~~~~~~~~~
/opt/compiler-explorer/libs/etl/include/etl/bit_stream.h:1034:5: note: candidate: 'etl::bit_stream_reader::bit_stream_reader(etl::span<char>, etl::endian)'
 1034 |     bit_stream_reader(etl::span<char> span_, etl::endian stream_endianness_)
      |     ^~~~~~~~~~~~~~~~~
```

See [this example on Compiler Explorer][godbolt] using:

- gcc 13.2.0
- ETLCPP 20.38.0
- C++98

[godbolt]: https://godbolt.org/#z:OYLghAFBqd5QCxAYwPYBMCmBRdBLAF1QCcAaPECAMzwBtMA7AQwFtMQByARg9KtQYEAysib0QXACx8BBAKoBnTAAUAHpwAMvAFYTStJg1DIApACYAQuYukl9ZATwDKjdAGFUtAK4sGISQCspK4AMngMmAByPgBGmMQgAMwapAAOqAqETgwe3r7%2BQemZjgJhEdEscQnJtpj2JQxCBEzEBLk%2BfmZctfXZTS0EZVGx8UkpCs2t7fldtpOD4cOVo8kAlLaoXsTI7BzmieHI3lgA1CaJbpgEtAD0MYQA%2BhPEmKwAdAjn2CYaAIL7h2OmDOFyutwUqUMHy%2BP3%2BZgODCOXlO5zcE3wgmhiW%2BfwBiKBILRBHQWCoWJx/z%2BNxuJwA8gBpEAnMEgEAQwyorzhAgADgeBBhf2YbHZOxOBEwEweXmQCBaD3QAE9hXhkGcAOxWP4nHUnbAAFRCDyE%2Bt%2B%2BoAkm4Hm5aZETb9IvqTpkAF6YfknBg%2BB4xRUShQggAiJ0S5y1v11zuajjVXMEfIIJ3uBAeLWITEVJgCFi9LB9fslWeD52LmpOGlUbhS5dUADFa6Qa9hvurA2HYZGDUb7RarTa7abHczrqz2QxOdyE18k48xxBk6niOnFat29rdV3jabe9bbfahyyQAvnq88y8mFhiDOCAp57PIQxG4fXHhDKzaIRrphV4lwyZW7CsLUnqABKIG0iBTKHmOqJoAwEwnHGvL8oKvzCpKkJiv6KZwVKMpysQCrKqwqoauGnaGluZqWruA4Ok6rruomub5v6QYhmuEYbpRPY0f2%2B5OkhCbXouy5Zjm3q%2Bv6Rbsf%2BFg1lWjYVvWSmqM2/5tr%2BHbcd2258Xug5OtBD6wQICFCSh2IiXOC5phmP7kTpVE7vxhnDrQrLHgQ55nq8l7Xrex4Pk%2BI4gC%2Bb4gB%2BBBfg5gEAbiVI0tgYEQVBoUwRcFkEI2LFSYW2KwuhorAth0qyvKNCqJg6BkdpOqbrxfYGfRzp4G6Hq5QWAYlhxWnrjqExMDGiGTh6tlLhm4mddJATFokpbyRWik1ipTYtppjn1TxelNXRB7pSZmWjdlnqSV105BYYd4pnZK6cRRunUbtAnuZ5s7eaeDznv5yaBfehghR5YUMPgEVRTFnEaYBiWgeBkGvWyh1uLhiZZTlZ3SQVQqsBhTBYZKOFmSm%2BEVXgVU1XJdV6ttT20S9jEdRjkrsaGfVcVtj0uc1Q5ZSJt1TUzCgyT1ckKdWykNmtUNsw9zn6XtRkHRyFwoyN8b8ujeZ5ULVmXQw12ifZ91OY1dNuYeXk%2BV9fnxAFBtjoDrLhX4kWfvQsW4vFlLe9DvwcOsHkcAEvB%2BBwWikKgnBuNY1jOps2zAvsPCkAQmj%2B%2BsADW/jqm8ZjqjyZgAJwBOqkhcDy6pcAAbFX%2BicJIIdpxHnC8AoIApKnYf%2B6QcCwEgaAsKkdDxOQlAD0P9AJMAXCJGYpBYAAbqqmAAGp4JgADutKpIwnDJzQtASsQbfzk39zMMQip77w58tIqtIxNomAONfpAD2wgi0gwtBX138%2BYDELwwA3BiFoG3bgvAsAsEMMAcQf98AvAcHgBekom6YCqsgLwEpX7cjqE3D8MR0yXw8FgJu3k8AsGvusKgBhgAKDXpvbeu8IEyEECIMQ7ApCsPkEoNQTddDdAMEYFAMdLD6DwDENusB0JhWuKQFBCRvJeAYBnVY6xUCpAaOAgAtOiEsphLDWELjyE42jDCZFMaoE4BirCWFMYGTckRaQPGUOBWs5oQjYAeAACWwL8BxIF7GOOcSaEIpiADqoCInoO8kwUxO90CGBjLwVACjiB4CwFIiA6w7DPwaC4EG0w/DdFCIsCoVQ9BFCyAIIplSMjVIYEMcpoxui5KQQIfoUxPAdD0G0honSFjlBGAkVp8xamjIGE04ZEgcnxx2DMuuQdG5/0jhwE4qgeRV20VXSQJxgDIDVDPXOJwIC4EICQM48IuCrF4J3LQ6wIBIE2AQVIWDR4QHHsPYgkQcacA2VsnZ1ihHABOUolRNzgj4CIOk9Aeh%2BBsNEOILh8KeEqHUH/ARpAN7plSGnAOnBg6kFDuHVZtIsGvMTKgKg6zNnbN2fsw5iRjkQA8IPL5lzEjXNuXizOIAAhmDeOqLo6oNCSDMAEHkXAzA8kSIkaQgcG6kEoVwDQKRiUpJbrYPQdz06LLMLwZVqqiVN1Wdyruaj5HxEyM4SQQA%3D%3D

dhebbeker, 2023-09-01:
I already reported this [here in Slack][slack] and I know that @jwellbelove has confirmed it. I just want to make sure the report does not get lost because of the Slack workspace plan in order.

Also this way others have a chance to avoid duplicate reports.

[slack]: https://etlcpp.slack.com/archives/C7T3YLH3L/p1692104462547079

jwellbelove, 2023-09-28:
Fixed 20.38.2

closed 2023-09-28 by jwellbelove