ETLCPP/etl #756: Parameter pack compilation error

jputcu, 2023-09-04:
Using `etl-20.38.0`, I get the following errors on arduino/avr-gcc 7.3
~~~~
include\etl\parameter_pack.h:141:74: error: template definition of non-template 'constexpr const size_t etl::parameter_pack<TTypes>::index_of_type<T>::value'
   constexpr size_t parameter_pack<TTypes...>::template index_of_type<T>::value;
                                                                          ^~~~~
~~~~

jwellbelove, 2023-09-04:
Does it compile if you delete the `template` in `index_of_type<T>::value`?

jwellbelove, 2023-09-04:
What C++ standard are you using?

jputcu, 2023-09-04:
I'm using C++17, seems like the same issue as #704

jputcu, 2023-09-04:
I'm currently on `20.37.2` which doesn't give the error.

jwellbelove, 2023-09-04:
I thought it may be a C++ version issue, but I think it may be a GCC version issue.

jwellbelove, 2023-09-04:
[etl-18-32-11.patch](https://github.com/ETLCPP/etl/files/12515644/etl-18-32-11.patch)

jputcu, 2023-09-05:
I get the same error. It keeps taking line 141. If I hardcode the other path, it compiles.
My arduino/avr-gcc compiler uses the following flags:
`avr-g++.exe -DETL_NO_CPP_NAN_SUPPORT -DETL_NO_HUGE_VAL_SUPPORT -DETL_NO_STL -DETL_TARGET_DEVICE_GENERIC -DETL_TARGET_OS_NONE -DF_CPU=14745600 -mmcu=atmega128 -g -funsigned-char -mrelax -ffunction-sections -fdata-sections -mcall-prologues -fno-rtti -fno-exceptions -fno-threadsafe-statics -Os -DNDEBUG -std=c++1z -Wall -Wextra`.

I also noticed the following change related to the `ETL_NO_CPP_NAN_SUPPORT`
~~~~
include/etl/profiles/determine_compiler_language_support.h:198:0: warning: "ETL_NO_CPP_NAN_SUPPORT" redefined
   #define ETL_NO_CPP_NAN_SUPPORT
~~~~

jputcu, 2023-09-05:
`20.37.3` also has the same `parameter_pack` issue.

jwellbelove, 2023-09-05:
I wonder if the `-std=c++1z` is the issue? The line in `parameter_pack.h` tests for full C++17 spec.
Does the error go away if you use `-std=c++17`, if available?

jputcu, 2023-09-05:
That flag is selected by conan to get C++17 on gcc 7.3. I think the compiler is not fully C++17 yet.
Small note: both the official Microchip and Arduino compiler use the 7.3 version, so I think the issue I'm facing will probably also appear on your Arduino package.

jwellbelove, 2023-09-05:
What is the value of `__cplusplus` for `-std=c++1z`?

jputcu, 2023-09-05:
~~~~
echo | ~/.conan2/p/b/micro197c0baa3ac02/p/bin/avr-g++  -dM -E -x c++ --std=c++1z - | grep cplusplus
#define __cplusplus 201703L
~~~~
C++17 gives the same value.

jwellbelove, 2023-09-06:
[etl-756.patch](https://github.com/ETLCPP/etl/files/12535580/etl-756.patch)

jwellbelove, 2023-09-08:
Does the above patch work for you?

jputcu, 2023-09-08:
I was planning to try it today. The problem is that the error only shows in my top level project, ETL is the lowest level conan package, so I will need to hack some to apply the patch.
Would it be better in the future to let the CI build the unit tests using the AVR compiler, they don't need to run, to catch these compilation errors?

jwellbelove, 2023-09-08:
All you would need to do is put together a minimal app that invokes the problem type definition.

jputcu, 2023-09-08:
I've been able to apply the patch and all mentioned warnings are gone.

jwellbelove, 2023-09-08:
I added a preprocessor condition that checks for GCC & versions before 8.x.x

jwellbelove, 2023-09-28:
Fixed 20.38.2

closed 2023-09-28 by jwellbelove