ETLCPP/etl #770: Outdated documentation for the unordered_map's operator[]

tocic, 2023-10-06:
[The doc](https://www.etlcpp.com/unordered_map.html) says there is a member `operator[](key_parameter_t key) const`, but I can't find it [here](https://github.com/ETLCPP/etl/blob/fe35751dcb3487e954b2822b6f22fe823f3a6aa9/include/etl/unordered_map.h), and my sample program [doesn't compile][1]:

```cpp
#include <etl/unordered_map.h>

int main() {
    const etl::unordered_map<int, int, 1> x{{1,1}};
    x[0];
}
```

> error: passing 'const etl::unordered_map<int, int, 1>' as 'this' argument discards qualifiers

[1]:https://godbolt.org/#g:!((g:!((g:!((h:codeEditor,i:(filename:'1',fontScale:14,fontUsePx:'0',j:1,lang:c%2B%2B,selection:(endColumn:1,endLineNumber:7,positionColumn:1,positionLineNumber:7,selectionStartColumn:1,selectionStartLineNumber:7,startColumn:1,startLineNumber:7),source:'%23include+%3Cetl/unordered_map.h%3E%0A%0Aint+main()+%7B%0A++++const+etl::unordered_map%3Cint,+int,+1%3E+x%7B%7B1,1%7D%7D%3B%0A++++x%5B0%5D%3B%0A%7D%0A'),l:'5',n:'0',o:'C%2B%2B+source+%231',t:'0')),k:50,l:'4',n:'0',o:'',s:0,t:'0'),(g:!((g:!((h:compiler,i:(compiler:g132,deviceViewOpen:'1',filters:(b:'0',binary:'1',binaryObject:'1',commentOnly:'0',debugCalls:'1',demangle:'0',directives:'0',execute:'0',intel:'0',libraryCode:'0',trim:'1'),flagsViewOpen:'1',fontScale:14,fontUsePx:'0',j:1,lang:c%2B%2B,libs:!((name:etl,ver:trunk)),options:'',overrides:!(),selection:(endColumn:1,endLineNumber:1,positionColumn:1,positionLineNumber:1,selectionStartColumn:1,selectionStartLineNumber:1,startColumn:1,startLineNumber:1),source:1),l:'5',n:'0',o:'+x86-64+gcc+13.2+(Editor+%231)',t:'0')),k:50,l:'4',m:50,n:'0',o:'',s:0,t:'0'),(g:!((h:output,i:(compilerName:'x86-64+gcc+13.2',editorid:1,fontScale:14,fontUsePx:'0',j:1,wrap:'1'),l:'5',n:'0',o:'Output+of+x86-64+gcc+13.2+(Compiler+%231)',t:'0')),header:(),l:'4',m:50,n:'0',o:'',s:0,t:'0')),k:50,l:'3',n:'0',o:'',t:'0')),l:'2',n:'0',o:'',t:'0')),version:4

jwellbelove, 2023-10-07:
I have updated the documentation

closed 2023-10-07 by jwellbelove

tocic, 2023-10-07:
Looks like the sentence _"Returns a reference or const reference to the indexed element"_ should be changed to _"Returns a reference to the indexed element"_.

jwellbelove, 2023-10-07:
Thanks, fixed.