ETLCPP/etl #779: hash.h: warnings produced with -Wfloat-equal

digimokan, 2023-10-30:
In hash.h, when compiling with -Wfloat-equal, the "v == -0.0f" on line 401 (and similar on lines 435 and 469) produce "warning: comparing floating-point with '==' or '!=' is unsafe."

Using `fpclassify` and `FP_ZERO` to fix this issue would work, but I think these are not portable. It looks like `signbit` is portable (assuming you can use `math.h` or `cmath`?), so maybe something with `signbit` and '>' or '<' comparisons would work?

```
if (!(v < 0) && !(v > 0) && signbit(v))
```

jwellbelove, 2023-11-15:
The issue is that fpclassify & signbit are not available for C++98 or C++03, which the ETL still maintains compatibility with.
I may have to disable the warning for those lines.

digimokan, 2023-11-16:
Ok, I didn't realize that. I'm not sure how to replace `signbit` portably.

jwellbelove, 2023-11-16:
I think the test `!(v < 0) && !(v > 0)` should be enough, as the effective check is "For any variant of zero, make it positive zero".

jwellbelove, 2023-12-05:
Fixed 20.38.8

closed 2023-12-05 by jwellbelove