ETLCPP/etl #783: Cannot insert existing value to full set

ismailtzn, 2023-11-02:
When I try to insert a value to full set, it always throws set_full error, even if the value was already in the set.

jwellbelove, 2023-11-03:
This may also be an issue for `etl::flat_set` and `etl::unordered_set`.

ismailtzn, 2023-11-04:
You are right, I will look them too 😄

ismailtzn, 2023-11-04:
I checked `etl::reference_flat_set`, `etl::flat_set` and `etl::unordered_set`, this issue does not occur for `etl::reference_flat_set` and `etl::flat_set`. For those two I added tests. I tried to fix `etl::unordered_set` also.

By the way, I couldn't see any contribution instructions, so I created a pull request directly. If you follow a particular way of accepting contributions, I'd love to do it your way.

jwellbelove, 2023-12-05:
Fixed 20.38.7

PR #782 (2023-11-24): Set is full error when adding existing item
When I tried to insert a value that was already in the set to a full set , it throws set_full error.
I believe it should let me try inserting an existing value. 
Resolves #783
closed 2023-12-05 by jwellbelove